olofk/fusesoc #349: Allow for targets without a tool

saw235, 2020-01-31:
I am trying to somehow integrate some existing aldeclint script into my local swervolf project without messing with the backend edalize. For example, I have the following
```yaml
targets:
  aldeclint:
    hooks :
      post_run : [run_aldec]
```
but no matter how I do it, I am restricted by the need for tools to be specified under the _aldeclint_ target

**ERROR: No tool was supplied on command line or found in 'swervolf' core description**

Is there some other way to do this?

olofk, 2020-02-15:
Hi,

I just started experimenting with adding a generic tool that makes it easier to hook into other toolchains. Check out the https://github.com/olofk/edalize/tree/generic_tool branch and let me know if there is something in particular you would need

GCHQDeveloper028, 2020-07-08:
I have a case similar to this, I have a generator to run that takes some VHDL files and generates C/C++ headers from the constants within the VHDL files (for software to interact with the FPGA image).
I'm currently having to provide both a toplevel & a tool:

```yaml
targets:
  default:
    generate: [export_headers]
    filesets: [fileset_a, fileset_b]
    toplevel: fake_toplevel
    default_tool: modelsim
```

would your proposed generic_tool work in this instance too? I only need to be able to perform `run --setup --noexport`, no build needs to take place just need to trigger the generator(s)