olofk/fusesoc #352: Is it possible to add relative paths in fusesoc.conf file

rahulraveendran15-coder, 2020-02-09:
Is  it possible to add relative paths in fusesoc.conf file?
I need to checkin thefusesoc.conf file to my master repository so that everyone  within the same project can use  the same .conf file without generating or adding by themselves while using fusesoc(They can directly fusesoc with target), no other setup liking adding library and all is not required.

olofk, 2020-02-15:
Yes. Relative paths are fine

rahulraveendran15-coder, 2020-02-17:
I have tried adding relative paths  like below
```
#!/usr/bin/env python3
import os
   root = os.path.dirname(os.path.realpath(__file__))
[library.checkcore]
location = root + "./"
sync-uri = ./
sync-type = local
auto-sync = true

```

But its not able to recognise this import os keyword and rest
Could you please help me on this

imphil, 2020-02-17:
fusesoc.conf is not a Python script, it's a config file. So please remove all python code.

`location = .` should be sufficient.

closed 2020-12-18 by imphil