olofk/fusesoc #355: Using parent directory (..) in files path results with run failures (except the first run).

m-kru, 2020-02-14:
I have created the repository for recreating the issue: [fusesoc parent directory (..) issue](https://github.com/m-kru/fusesoc_issue).
Even if you do not have GHDL installed it is very easy to analyze. 
When I run the `tb_` for the first time it ends with success. However after the first run it always fails with following traceback:
```
INFO: Preparing ::ent:0
Traceback (most recent call last):
  File "/home/mkru/.local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc==1.9.4.dev30+gf59066f', 'console_scripts', 'fusesoc')()
  File "/home/mkru/.local/lib/python3.8/site-packages/fusesoc-1.9.4.dev30+gf59066f-py3.8.egg/fusesoc/main.py", line 837, in main
    args.func(cm, args)
  File "/home/mkru/.local/lib/python3.8/site-packages/fusesoc-1.9.4.dev30+gf59066f-py3.8.egg/fusesoc/main.py", line 355, in run
    run_backend(
  File "/home/mkru/.local/lib/python3.8/site-packages/fusesoc-1.9.4.dev30+gf59066f-py3.8.egg/fusesoc/main.py", line 423, in run_backend
    edalizer = Edalizer(
  File "/home/mkru/.local/lib/python3.8/site-packages/fusesoc-1.9.4.dev30+gf59066f-py3.8.egg/fusesoc/edalizer.py", line 66, in __init__
    core.export(files_root, _flags)
  File "/home/mkru/.local/lib/python3.8/site-packages/fusesoc-1.9.4.dev30+gf59066f-py3.8.egg/fusesoc/capi2/core.py", line 208, in export
    os.makedirs(os.path.join(dst_dir, d))
  File "/usr/lib/python3.8/os.py", line 221, in makedirs
    mkdir(name, mode)
FileExistsError: [Errno 17] File exists: '/home/mkru/workspace/fusesoc_issue/build/ent_0/src/ent_0/../tb'
```

The issue is caused by referencing the parent directory (..) in ent.core file. When I move ent.core one directory up and remove `../` from file paths I can run `tb_` target multiple times.

I know that I can remove `build` directory each time I run target, however this approach has 2 drawbacks:

1. It is annoying and time consuming.
2.  In case of multiple cores and automated tests only the first one will succeed. Assuming each core has its own directory and within each core directory there is the same structure of subdirectories - `rtl`, `tb`, `fusesoc` directories.

I am also aware that I can just move the `.core` files one directory up, however right now I work on a project where keeping `.core` files in a particular `fusesoc` directory is more convenient.

imphil, 2020-02-14:
Paths starting with `..` are not supported in fusesoc. Unfortunately, there is no warning in place at the moment, getting this implemented is tracked in https://github.com/olofk/fusesoc/issues/264.

Simply place the core file in the top-level directory and everything should work as expected.

m-kru, 2020-02-14:
Is it hard to implement, or is there any other problem?

imphil, 2020-12-18:
I honestly don't recall the exact details right now, but it was effectively very hard to get right as it breaks the assumption that cores are "self-contained" within a directory. This assumption is important when e.g. copying source files to the build directory, when using libraries (and ensuring that cores don't "escape" a library), etc. Putting the core file into the root of the IP block is generally easy enough and works reliably. Please let us know if you come across a use case where that's not true, and we'll try to find a specific solution.

closed 2020-12-18 by imphil