olofk/fusesoc #356: ghdl: compile order

csantosb, 2020-02-15:
Hi Olof,

  I have started playing around with fusesoc.

  Firts, my own lib: [ip-lib](https://github.com/csantosb/Ip-Lib). Then, a simple [ip](https://github.com/ip-vhdl/Moving-Window) to simulate (ghdl) and implement (ise and vivado). As you may see, only a few rtl files are necessary here.

  So far, so good, except the order of .vhd files in the .core file counts for simulation. Swapping two of the rtl files makes simulation fail.

  Looking closely, I observe that the resulting Makefile produced by edalize contains the following lines for analize target

    	ghdl -a $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/top_pkg.vhd
	    ghdl -a $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/averager.vhd
	    ghdl -a $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/top.vhd
	    ghdl -a $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/sim/runs/averager/hdl/averager_tb.vhd

and 

    ghdl --elab-run $(ANALYZE_OPTIONS) $(STD) $(TOPLEVEL) $(RUN_OPTIONS) $(EXTRA_OPTIONS)

for run target. This is in agreement with ghdl online doc at [quick start guide](https://ghdl.readthedocs.io/en/latest/examples/quick_start/README.html).

Now, as stated in [Working with non-trivial designs](https://ghdl.readthedocs.io/en/latest/examples/quick_start/DLXModelSuite.html), another approach is available: importing sources all together. Thus, whether you 

    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/top_pkg.vhd

(same for remaining rtl sources), whether

    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/*.vhd

and then you `ghdl -m` and `ghdl -t` the top level. This way, the order of rtl sources in .core file becomes irrelevant (as under vivavo or ise).

  I hope my suggestion its clear: replacing all `ghdl -a` by a `ghdl -i` improves things. 
  
  As an aside, related question, is it possible to include globbing in the .core yaml file ? Providing the path to a folder with all source rtl would be great, too.

  Anyway, thanks for this great tool, I really like it !

olofk, 2020-02-15:
Hi Cayetano,

Happy to hear you like it. Thanks for letting me know.

This sounds like a good idea. FuseSoC guarantees to keep the compile order that is specified in the core description files, but sometimes it can be tricky to specify the correct order, so if the `-i` switch improves the situation I think we should use that.

I don't want to add globbing though. I agree that it makes it a bit quicker to set up things but it also makes it harder to track which files to use. In the future I want to make it easier to just recompile the parts that have been changed and having an exact list of files makes it easier to keep track of that.

csantosb, 2020-02-15:
Fine with me then, thanks. 

Don't forget updating the `--elab-run` flag too: as long as I understand, two stages are needed in this case, `ghdl -m` followed by `ghdl -r` (and not `ghdl -t`, a typo in my prev message).

olofk, 2020-02-15:
That's good to know. @tgingold, do you have any input on which alternative to prefer? Just in case I'm missing something

tgingold, 2020-02-16:
Should the source files in a .core be ordered ?  I fear that if the source files aren't ordered, you will have issues with any simulator and find a way to automatically order them.

If you don't want to have them ordered, yes using `-i`, then `-m` and then `--elab-run` is the way to go.

csantosb, 2020-02-16:
So what about something like this ?

    #Auto generated by Edalize
    STD = --std=08
    TOPLEVEL = averager_tb
    ANALYZE_OPTIONS = --ieee=synopsys -C --work=work -g -O2 -P/home/csantos/CompiledLibraries/Vivado --mb-comments --warn-unused -fexplicit -frelaxed-rules --no-vital-checks -fpsl
    RUN_OPTIONS = --wave=wave.ghw --ieee-asserts=enable  --stop-time=100us

    VHDL_SOURCES = ../src/moving-window_0/sim/runs/averager/hdl/averager_tb.vhd ../src/moving-window_0/src/top_pkg.vhd ../src/moving-window_0/src/top.vhd ../src/moving-window_0/src/averager.vhd

    all: run

    run: $(TOPLEVEL)
	    ghdl -r $(TOPLEVEL) $(RUN_OPTIONS) $(EXTRA_OPTIONS)

    work-obj08.cf:
	    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/sim/runs/averager/hdl/averager_tb.vhd
	    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/top_pkg.vhd
	    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/top.vhd
	    ghdl -i $(STD) $(ANALYZE_OPTIONS) ../src/moving-window_0/src/averager.vhd

    $(TOPLEVEL): $(VHDL_SOURCES) work-obj08.cf
	    ghdl -m $(STD) $(ANALYZE_OPTIONS) $(TOPLEVEL)

Provided you construct the `VHDL_SOURCES` array and considers the standard, this is just a `make run` away from running the simulation. If one decides to play with the rtl code under the `build/sim-ghdl`, same command updates and runs the executable, no need to build the library back again. Just tested with this IP and seems ok to me.

tgingold, 2020-02-16:
Yes, that's fine.  You can also use `ghdl -i` with all the source files instead of calling it for each source file.

csantosb, 2020-02-16:
Just checked

    work-obj08.cf:
	    ghdl -i $(STD) $(ANALYZE_OPTIONS) $(VHDL_SOURCES)

Does it too.

ThomasHornschuh, 2020-02-17:
Just a comment for my side as happy FuseSoC /ghdl user.
I'm in favor with tgingold's recommendation:
> If you don't want to have them ordered, yes using `-i`, then `-m` and then `--elab-run` is the way to go.
This would make sure that it runs with ghdl mcode and llvm/gcc backends. 
I use mcode backend in most cases, because it is very fast in analyzing/compiling which often outweighs the slower simulation runtime. 

It would be helpful when  FuseSoC would not hardcode "ghdl" in the Makefile, but would use a variable for this e.g. $(FUSESOC_GHDL). There is no need for further support in FuseSoC to set such a variable, setting it over the environment would be fine.

csantosb, 2020-02-29:
Fixed in [here](https://github.com/olofk/edalize/pull/114).

closed 2020-02-29 by csantosb