olofk/fusesoc #360: Improve wording regarding the core file.

secworks, 2020-02-23:
Currently the wording is:

"First thing to do is to create a new directory and put a file inside with a .core. We will use the directory ~/fusesoc_tutorial/cores/gpsemulator and name the file blinky.core."

At least to me this is a bit confusing. I would suggest change to "with a .core suffix". Also one has to read the second sentence to understand what "put a file" means. Something like "Create a file with the name of the core and a .core suffix" may be easier to understand.

imphil, 2020-02-23:
Good idea! Would you mind doing a pull request with the suggested change?

secworks, 2020-02-25:
Sure, I'll create a patch for it.

imphil, 2020-12-18:
The documentation that contained this sentence is significantly reworked now, let's assume this fixed. Please let us know if things are still unclear by reopening this issue (or filing a new one).

closed 2020-12-18 by imphil