olofk/fusesoc #361: Remove CAPI1 support

imphil, 2020-03-02:
The CAPI1 support is getting less and less testing, and the code paths are tricky to maintain. I'm opening this issue to see how we can deprecate and then at a later stage remove CAPI1 support from fusesoc.

Plan:
1. Warn if CAPI1 cores are used. Do that for a bit of time.
2. Remove CAPI1 support. Do a new major release (e.g. fusesoc 2.x).

Ideally, we also have #188 (with work started in #231) ready. But it turns out that's not super simple to do, with unknown benefit. (We don't know how many users are actually using CAPI1 files these days.) So I'd not make that a dependency for removing CAPI1 support unless someone complains very loudly.

WDYT @olofk ?

olofk, 2020-03-03:
This has been on my mind for a long time. I agree with your item list, but would like to add two more items before them
-1. Deprecate `orpsoc-cores` for real by rewriting `fusesoc init` so it doesn't try to install it
0. Remove all CAPI1 cores from `fusesoc-cores`. If there is a version with CAPI2, just drop the older ones. Otherwise create a new CAPI2 version

ThomasHornschuh, 2020-03-03:
As a happy FuseSoC user with ~20 own cores I hope that you have some automated tool to convert CAPI1 to CAPI2 before removing CAPI1 support.  I'm aware that converting my cores is long overdue and assume that it will not take more than 1-2 days, but I'm always a bit reluctant to spend time for changes that does not give me an immediate added value in the moment I'm doing it.  In sum over all tools I'm using such things add up to more time I like to spend on it.

I assume that FuseSoC has an uniform in-memory representation of the core description. So I would find  some option like "write-corefile" which just write this representation into a CAPI2 the most convenient.

imphil, 2020-03-03:
There is unfortunately no "uninform in-memory representation", so a converter is significant work to be usable in a general case. If you want to contribute to that, there's a starting point in  #231.

ThomasHornschuh, 2020-03-03:
A short look into the PR shows that #231 seems to be a stand alone program. 
@olofk: What do you think is a better starting point, standalone program or add the functionality into FuseSoC?  
My understanding of FuseSoC code is that capi1 and capi2 be two Independent code paths. 
Eventually the capi1 path can become a "conversion only"  functionality. So a user can list and convert capi1 cores with it,  but not run them in a future version. 

Maybe to goal should be that the converter can handle "no-frills" simple cores and for all other be able to build a valid skeleton, containing dependencies, filesets, targets. 

I cannot promise anything, but at least I will run #231 over some of my cores.

olofk, 2020-03-03:
The CAPI1 support is pretty well compartmentalized so it doesn't hurt the code base too much. At the same time I think there are some simplifications that can be made to the code base once it's removed, and it's an absolute requirement to have it gone once we start looking at a CAPI3 (which is not yet).

The current code implements an internal API against CAPI1 and CAPI2 cores to create the EDAM file. So there is no direct conversion between the CAPI types. So I don't think we can get much help from the current code base. CAPI1 and CAPI2 are also conceptually pretty different so there is no 1-to-1 mapping between them. I think a standalone tool that can do one-time conversions for most cores with some manual fixups would be the easiest way forward. I don't think there are enough CAPI1 cores to justify something very complicated.

olofk, 2020-03-05:
I have started cleaning out CAPI1 cores from fusesoc-cores but there is still a bunch of them left

benreynwar, 2020-03-05:
Just checked my repo at work.  It has 321 CAPI1 core files, so it looks like I'll be testing out the conversion script soon :).

ThomasHornschuh, 2020-03-07:
I have started to test the conversion tool on my first cores yet (only comparable simple ones which a few files). These cores have only simulation targets, because they are peripheral cores not to be intended to be synthesized stand alone. All are VHDL based and simulated with GHDL.

My first experience/observation so far: 

- The conv option differs from other fusesoc options that it expects to specify a core file, not a core.
- The --output options seems to be mandatory to create a CAPI2 file, if ommited nothing happens
Example command line:
`fusesoc --config fusesoc.conf conv bonfire-soc-io/bonfire-soc-io.core --output=bonfire-soc-ioo/capi2.core`
This will create a file capi2.core from the file at bonfire-soc-io/bonfire-soc-io.core 

For my simple cores the capi2 file was valid and worked. Additional observations:

- The name: section is placed somehwere in the midle of the file, still valid, but I would prefer it at the top.
- Dependencies are copied to all file sets. I think this cannot be avoided because of the semantic change from dependencies on cores to dependencies on file sets. 
- Scope = private from CAPI1 seems to be translated correctly by omitting the fileset from the default target.

You can find an example here: https://gist.github.com/ThomasHornschuh/a3c5c8a4cca2409d7f1505db0864e3b9

imphil, 2020-12-18:
Thanks @ThomasHornschuh for trying it out; I've done similar test and I think the tool is good enough to give people a starting point for their conversion. So we have a tool that gets us 90% there in the conversion, which is already quite significant IMO. I've updated the PR at #231 to improve the input/output command line handling and suggest we just take it like that. It is quite impressive work, so thanks a lot @Oxore!

PR #457 (2021-03-22): Remove CAPI1 support
- Remove all CAPI1 support code.
- Leave a parsing warning in place if a CAPI1 core is found.
- Convert all in-tree core files to CAPI2.
- Remove unused core files.

Note: ipxact support was CAPI1-only and is now removed as well.

Fixes #361
commit 29f43d1d (2021-03-22): Remove CAPI1 support

- Remove all CAPI1 support code.
- Leave a parsing warning in place if a CAPI1 core is found.
- Convert all in-tree core files to CAPI2.
- Remove unused core files.

Note: ipxact support was CAPI1-only and is now removed as well.

Fixes #361
closed 2021-03-22 by olofk