olofk/fusesoc #363: Rewrite of argument parsing

olofk, 2020-03-04:
I'm working on rewriting the argument parsing. Since this is a fairly long process and involves at least @imphil and @kgugala and probably others I wanted to write it down here so we can discuss it in public.

The maon goal is to get rid of argument parsing in Edalize and move it into FuseSoC. Arguments are currently parsed at two points in Edalize (`configure` and `run`) which will be handled in the following way:

`configure`: FuseSoC (or any other applicable frontend) will parse the arguments and write the parameter values to the `default` key for each parameter. This makes it no longer necessary to parse arguments on the Edalize side

`run` : FuseSoC (or any other applicable frontend) will parse the arguments and send a dict of key/value pairs to Edalize. Edalize will update its (run-time) parameter values accordingly (This also gives edalize the possibility to check/warn if any build-time arguments were changed without a rebuild)

Here's the full plan
1. Get rid of the `cli_parser` special case for verilator by introducing an extra_args/runtime_args argument in EDAM to allow passing arguments to verilated models that aren't parsed by FuseSoC/Edalize and deprecate the `cli_parser` option. Update FuseSoC migration guide
2. Add argument parsing to FuseSoC to set parsed values in EDAM file. Send an empty argument list to Edalize.configure
3. Release a new FuseSoC version 1.10
4. Make Edalize changes
4.1 Add a setup.py blocker on FuseSoC<1.10
4.2 Drop argument parsing from configure
4.3 In Edalize.run, check if incoming arguments is a list or dict. If list, run parse_args. If dict, assume key/value pairs and update internal parameter values.
4.4 Update example in README.md
4.5 Release edalize 0.2 and communicate API break
5. Back in FuseSoC, depend on edalize >= 0.2 and send parsed key/value pairs to edalize.run

olofk, 2020-03-04:
FYI, I have already implemented 2 but not pushed it yet

olofk, 2020-03-04:
1 is pushed to Edalize now. Added a warning instead of updating the migration guide

olofk, 2020-03-04:
Pushed 2 now #yolo #carpediem

olofk, 2020-03-04:
3 done. Time to work on the Edalize side

DHWinterWolf, 2020-03-22:
I think the issue I originally thought was with Edalize, https://github.com/olofk/edalize/issues/126 is actually related to this. The parsing seems to be done well by FuseSoC, so doesn't need to be re-done by Edalize. I assume step 4 would remove this, but the patch I'm proposing to FuseSoC in https://github.com/olofk/fusesoc/pull/380 stops the break until that happens, I think.

This is well worth others look at, though: whilst I've done some local testing, I don't pass many arguments to the run step, so I might have missed a use case.

olofk, 2020-04-02:
So things are still in flux. FuseSoC still leaves the parsing to edalize when we run without configuring first. Thinking out loud how to get this finished up

backend.run(args) wants a dict of all (run-time) parameters
We only care about arguments specified on the command-line
To parse the command-line arguments we need to build a parser
This parser needs to know all valid arguments.
Here's the problem. How do we get all valid arguments?

We can either...

..rebuild the dependency tree, create the EDAM struct and get it from the "parameters" key (like we do in configure). But this feels clumsy and what if we for some reason end up with a different dependency tree?

...or...

...parse the EDAM file that should be in the work root already. This must be the right answer!! We construct a new parser based on the current backend_class and the parameters section of the EDAM file

DHWinterWolf, 2020-04-03:
I'd assumed that the EDAM file was re-parsed in the run section, that's my bad, sorry. I'd agree that this sounds like the right answer.

olofk, 2020-05-27:
Finally gotten around to fix this. Please reopen if there are still issues remaining

closed 2020-05-27 by olofk