olofk/fusesoc #371: Generic targets

csantosb, 2020-03-10:
Hi Olofk,

  I'm trying to simulate [this](https://gitlab.com/ip-vhdl/mux_fifo/-/tree/master/sim%2Fruns%2Fmux_fifo.cocotb) ip, using cocotb. To that purpose, I'm using the provided, customized makefile.

  Now, cocotb is not yet part of the targets. Would it be interesting to provide a generic target, not related to any particular tool ? In this case, no Makefile is build for any tool, and the user is responsible and assumed to provide all necessary files, as usual, but also de Makefile : in my case, a Makefile, python module, a hdl wrapper and anything else. Fusesoc just calls the Makefile, as usual.

  What do you think ?

imphil, 2020-03-10:
I would suggest writing a normal edalize backend for cocotb, or is there anything which prevents you from going down that route?

csantosb, 2020-03-11:
For sure, having a cocotb backend might be useful in general. But, as you may see, this use case implies a customized Makefiles, departing from the cocotb usual way of doing things: I declare custom variables, I include another Makefiles, and so on. So here we are speaking about custom, general purpose and user related functionality rather than a standard tool flow. For example, one might provide a Makefile for creating docs, doing several simulations in different languages, or producing any outcom related to a collaboration or project. Anything which departs from a usual tool flow. In my case, I try to include a more complex Makefile than the one provided by edalize, even if a cocotb backend would exist.

olofk, 2020-03-16:
I have started work on a generic tool [here](https://github.com/olofk/edalize/tree/generic_tool). I agree with @imphil that a cocotb backend would be very good to have, but there have been enough people to convince me that a generic one would be handy. I will close this one though, because this issue should be in Edalize instead

closed 2020-03-16 by olofk