olofk/fusesoc #378: fusesoc run --elf-load error (ERROR: Setup failed : Unknown parameter elf_load)

birukgit, 2020-03-18:
Dear olofk,

I have been trying to work with fusesoc for mor1kx openrisc cpu. While running the fusesoc run command I am getting this error message shown below.
ERROR: Setup failed : Unknown parameter elf_load
Could you please help me how to solve this issue.

Jbalkind, 2020-06-11:
I'm also perplexed by this. I can't see anything mentioning elf_load in any of the sources put in build/src/. The parameter used is --elf-load which is correct according to the elf-loader core. Is the error in another CAPI file? How do we debug that if they are remote cores?

olofk, 2020-06-16:
I might have an idea what's going on. Have an old memory of something similar. It could be that the dash gets converted to an underscore somewhere in the code, because python's argparse module does that internally. I don't have time to look into this properly right now but my suggestion would be to trace it through the code and see if and where it gets converted between  `elf-load` and `elf_load`

Jbalkind, 2020-06-16:
Do you recall a particular reason for this line?

https://github.com/olofk/fusesoc/blob/6f90890635fdc6bc34e1bb1e0f2b3120a1ddb109/fusesoc/edalizer.py#L295

Jbalkind, 2020-06-16:
Ok I can't really see a way around argparse changing - in arguments to _ but that seems to suggest that parameters can't have - in them? There must have been a change to this at some point because elf-loader does indeed use that elf-load parameter (not elf_load) and seems to have worked at some unspecified point in the past, though it is a CAPI1 core. Changing elf-load to elf_load in the core itself does let things continue.

olofk, 2020-06-16:
That line is probably to match the renaming that argparse does internally. IIRC the problem used to be the opposite, namely FuseSoC changed all _ to - in an attempt to undo the damage done by argparse. My memory is a bit hazy here but either I missed this corner case or I considered it to be the least destructive option. Going forward it could be easiest to just outlaw dashes in parameters. Not even sure all EDA tools properly handle dashes.

Let's begin by changing the --elf-load to --elf_load and sweep it under the carpet for a bit longer

0ena, 2020-07-21:
Hi,

I am a new user of OpenRISC (or at least I am trying to be :-) ).
I am as well trying to run a simple "Hello world" in the Instruction Set Simulator of mor1kx using the following command in fusesoc:
`fusesoc sim mor1kx-generic --elf-load hello.elf`

I do also get the following message:
"ERROR: Setup failed : Unknown parameter elf_load"

this error remains even if I change the command to
`fusesoc sim mor1kx-generic --elf_load hello.elf`
 
as suggested above. 
Am I doing something wrong? Is there a fix or workaround to make it work?
I am using Ubuntu 18.04.4 LTS. 

Thank you in advance for your time and response!

P.S: my intention is to run at some point the vmlinux elf.

Jbalkind, 2020-07-21:
If you're following these excellent instructions as I was: https://stffrdhrn.github.io/hardware/embedded/openrisc/2019/06/11/or1k_marocchino.html

Then in my case I was able to clone the elf-loader core directly from here: https://github.com/fusesoc/elf-loader

Then `fusesoc library add elf-loader /path/to/elf-loader` to use that local copy rather than the remote one
And modify elf-loader.core to change the parameter to elf_load instead of elf-load

From there I think the commands you used worked just fine for me

0ena, 2020-07-21:
Worked like a charm!
Thank you very much for your help on this @Jbalkind !

I am trying to build an implementation of an OpenRISC on a DE0-CV board and the different documentation/tutorials/links all over the internet make this pretty challenging. :-)

stffrdhrn, 2020-10-09:
I ran into this today after upgrading fusesoc.  I created a PR for elf-loader as per @Jbalkind 's suggestion.  The tutorials will have to be updated to `--elf_load` too.  

https://github.com/fusesoc/elf-loader/pull/2

imphil, 2020-12-18:
Looks like there's nothing to be done on the fusesoc side for this issue?

Jbalkind, 2020-12-18:
Might be worth having some constraint on the names of flags in CAPI to avoid this? I can't remember whether this was purely CAPI1 specific, though.

stffrdhrn, 2020-12-26:
I forgot about this one. I think think we weren't planning to fix anything in fusesoc. It would be nice to just support --elf_load and --elf-load and --elf_any-other_combination transparently.