olofk/fusesoc #381: [doc] Document use flags

imphil, 2020-03-23:
FuseSoC supports use flags, but doesn't document which use flags are actually set. This needs to be documented.

Currently we set:

- A target use flag target_TARGETNAME, e.g. `target_sim` if fusesoc is called with `--target=sim`.
- A tool use flags tool_TOOLNAME, e.g. `tool_verilator` if fusesoc is called with `--tool=verilator`

Support for user-defined use flags is being developed in #268, but currently not present.

olofk, 2020-03-26:
A rudimentary first version of user-defined use flags are actually implemented. It's possible to specify --flag=i_am_a_flag --flag=i_am_another_flag on the command line

olofk, 2025-02-21:
An amazing engineer actually added documentation for this some time ago. https://fusesoc.readthedocs.io/en/stable/user/build_system/flags.html#built-in-flags

closed 2025-02-21 by olofk