olofk/fusesoc #383: Support paramtype "generic" for backend Vivado

ThomasHornschuh, 2020-03-31:
Not sure of this is an edalize instead of fusesoc issue:
When using VHDL based cores with Vivado/xsim paramtype generic is not supported and lead to an error message:
`WARNING:root:Parameter 'TestFile' has unsupported type 'generic' for requested backend `

Workaround at the moment is using "vlogparam", but then I get a depreciation warning when using ghdl. 

BTW: In addition it would be good to support the real datatype for VHDL generics.

GCHQDeveloper560, 2020-04-07:
Lack of `xsim` support for generics is an Edalize issue. Generics have been supported by the Vivado synthesis/place-and-route backend (`vivado`) for some time, but you're correct that they are not supported for the Vivado simulation backend (`xsim`). I've hopefully correctly added support in [my xsim_generic branch](https://github.com/GCHQDeveloper560/edalize/tree/xsim_generic).

Are you able to test this and see if it works for you? Hopefully installing that branch is as simple as
```
pip install -U git+https://github.com/GCHQDeveloper560/edalize.git@xsim_generic
```

Support for a floating point data type is a (separate) FuseSoC issue.

olofk, 2020-04-07:
@GCHQDeveloper560, your xsim_generic branch looks fine to me from a quick glance. Happy to let it go in if it seems to work.

Handling floating point natively is probably a good idea. I have used strings to pass floats before (e.g. [here](https://github.com/olofk/stream_utils/blob/master/stream_utils.core#L41). That worked with plusargs in icarus but it probably breaks for backends and parameter types that wants to quote strings but not floating points.

If it's possible to pass them as strings for now, that would be great. Otherwise we need to look into that eventually

GCHQDeveloper560, 2020-04-08:
I took a look at the FuseSoC end and unless I missed something it's a pretty simple addition. See (my real_genparam branch)[https://github.com/GCHQDeveloper560/fusesoc/tree/real_genparam]. Rather than me dreaming up a bogus core/HDL as a test does this work for your case @ThomasHornschuh?

ThomasHornschuh, 2020-04-09:
I tried you edalize version. Unfortunately it screwed everything up. I got the traceback:
````
File "/usr/local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/home/thomas/development/fusesoc/fusesoc/main.py", line 830, in main
    args.func(cm, args)
  File "/home/thomas/development/fusesoc/fusesoc/main.py", line 365, in run
    args.build_root,
  File "/home/thomas/development/fusesoc/fusesoc/main.py", line 456, in run_backend
    backend.configure([])
  File "/usr/local/lib/python3.5/dist-packages/edalize/edatool.py", line 112, in configure
    self.configure_main()
  File "/usr/local/lib/python3.5/dist-packages/edalize/ghdl.py", line 109, in configure_main
    standard=stdarg[0].split("=")[1],
IndexError: list index out of range
````
on all my cores. Unfortunately my complete fusesoc setup is now broken, no idea why. I tried to install standard edalize again (tried several methods). The message persists.

I'm not proficient enough with python/pip and all these stuff to find out what is wrong. Hope that somebody can help me. I suspect that there is some python 3.5 / 2.7 version mixup (I'm on Ubuntu 16.04 LTS)
I have installed fusesoc and edalize from the git repos with pip -e. 

Anyway, maybe a good opportunity to continue the setup of my new development VM...

ThomasHornschuh, 2020-04-09:
Update: 
Fortunately I wrote down the edalize version I was using before, it is 0.1.7
Manually checking out this version and pip -e on it fixed everything. 
My versions of fusesoc and edalize are:
````
edalize                      0.1.7                  /home/thomas/development/edalize
fusesoc                      1.11.dev27+ged09c2e    /home/thomas/development/fusesoc
````
With edalize 0.2.0 the stack trace above comes with all my (ghdl) cores. Not tested yet with xsim.
It seems not dependent on a specific core, so I suspect some problem with my individual setup.

ThomasHornschuh, 2020-04-10:
Finally resolved what is going on (after a night of sleep :-) )
The stack trace happens because "std" parameter is not set when there is no version specific vhdl file_type is set.
Another problem is that the ghdl command line did not work when ghdl mcode is used. With mcode backend ghdl elaborates and simulates in one step, for this reason the elaboration options must also be passed to the -r command.
According to ghdl doc the syntax of the -r command is:
`-r <[options...] primary_unit [secondary_unit] [simulation_options...]>`
The older ghdl.py from edalize 1.7 was correct in this respect, it also used the --elab-run command. 

To my understanding --elab-run automatically does the steps which are needed for different ghdl backends, so it hides the differences.
`ghdl -m followed by ghdl -r` works also for mcode, the -m step is just superflous, but does not harm.

In general I find the -std determination a bit strange, I would prefer setting the -std option in a direct way (or at least allow overwriting of the determination logic). A additional pitfall is, that the is no file type "vhdlSource-93c". So setting vhdlSource-93 will set `--std=93`, which will often fail, because most code will expect the `--std=93`c default of ghdl.

ThomasHornschuh, 2020-04-10:
The issue in ghdl.py has no relation to the xsim_generic branch. The patch in xism_generic works perfect with xsim. The failure with ghdl was a side effect of the edalize update. I have created a PR for edalize.

GCHQDeveloper560, 2020-04-14:
Glad to hear of the good results with generics in xsim. I've put in a PR. Were you able to test real generics/parameters with FuseSoC?

ThomasHornschuh, 2020-04-18:
Yes. My patched edalize version and your patches together make it working. My cores now use generic as Parameter type and can be simulated with ghdl and xsim (and build with Vivado also)

GCHQDeveloper560, 2020-04-20:
Does [this branch to add proper real support](https://github.com/GCHQDeveloper560/fusesoc/tree/real_genparam)  rather than the string workaround work for GHDL? It doesn't do anything special to the real values. We've had cases with types like boolean where we've had to mangle the values for different tools, so hopefully real isn't in the same boat.

DHWinterWolf, 2020-06-16:
I've now tested that branch with ModelSim, if that helps. The result is interesting — it does indeed provide a method that works nicely. However, the result is exactly the same as the current FuseSoC/Edalize versions do anyway. It looks like your branch was taken before the latest releases, when changes were made to the way parameters are parsed. This also explains why your code doesn't merge cleanly into master at present.

To summarise, from a ModelSim point of view, there's no difference in the output of latest master FuseSoC when setting a parameter of type "str" compared to your branch and using a parameter type of "real".

The parameter handling has been simplified a fair bit in recent releases, I wonder if GHDL might also work with type "str" for VHDL real generics. I'm currently tempted to just drop the type from my core file and let it pick up the default, which looks like it'll work for ModelSim at least.

GCHQDeveloper560, 2020-06-16:
It does look like the CAPI 2 parameter handling code has changed quite a lot. I've rebased and fixed the conflicts in the branch.

I'm not surprised ModelSim is willing to set a real from a string. Hopefully there's not as much variation in what tools expect as with boolean! If real gives the same result as the string workaround I'll consider that a success and submit a PR. Hopefully others can point out any quirks with GHDL or other tools.

DHWinterWolf, 2020-06-16:
I've just re-tested the rebased one, and again everything looks fine, thanks for your help with this.

From experience, ModelSim seems to accept anything that would be considered valid if it was included in a generic map, pretty much as if -g<x>=<y> means that <y> is pasted into a generic map verbatim and then elaborated. I've nothing other than empirical evidence to back that up, though!

olofk, 2020-06-16:
I realized the conversation diverged somewhat from the topic, but FYI I just pushed #411 to support real parameters. Thanks everyone for working on that.

But I lost track of if generics are supported in vivado now. Can someone confirm and close if so?

GCHQDeveloper560, 2020-06-16:
The xsim generic support was merged in olofk/edalize#137, so I think this can be closed. It would be interesting if @ThomasHornschuh could test the real support with GHDL. He had reported problems above, but I think another option could be opened for any problems there.

olofk, 2020-06-16:
Yes. This is getting a bit overloaded, so let's close and open new ones that are more on-topic

closed 2020-06-16 by olofk

ThomasHornschuh, 2020-06-16:
> The xsim generic support was merged in [olofk/edalize#137](https://github.com/olofk/edalize/pull/137), so I think this can be closed. It would be interesting if @ThomasHornschuh could test the real [...]

Just confirmed that it works for me with ghdl and Vivado (xsim).