olofk/fusesoc #393: pre_run hooks do not trigger when --run argument is passed

Bucknalla, 2020-04-15:
It's not clear if this is intended behaviour but I believe that the given a command such as `fusesoc run --run --tool=vivado --target=ultra96_v1 my_core`, that my `pre_run` hooks should still be observed and triggered according to the hooks outlined in the target `ultra96_v1`. For example:

```
targets:
  default:
    filesets : [rtl]
    toplevel: ["is_toplevel? (base_wrapper)"]
  ultra96_v1:
    default_tool: vivado
    description : Ultra96v1 ZynqMP Board
    filesets : [ultra96v1]
    hooks: 
      pre_run: [my_pre_run_hook] <--- Does not trigger when '--run' is passed
    tools:
      vivado:
        part : xczu3eg-sbva484-1-e
    toplevel : ultra96_wrapper
```

This is seemingly ignored when the `--run` argument is passed into the `fusesoc run` command.

Let me know if I'm missing something there!

Thanks!

Bucknalla, 2020-04-22:
It looks like this requires the `--setup` arg to be passed in order for the hooks to run?

imphil, 2020-04-22:
Thanks for your issue. This looks like a bug. I didn't have time to fully investigate yet, but looking at the code I don't see it immediately. Hooks are executed here: https://github.com/olofk/edalize/blob/454760bf6f7fcbf1324f8e02fcd3c038516accc0/edalize/edatool.py#L141

* Can you have a look at the generated eda.yml file in your `build` directory and paste the hooks part into this issue for us to have a look?
* Can you also ensure that you're using the latest edalize version, by using `pip install -U edalize`?

olofk, 2020-06-16:
Sorry. I haven't had time to look at this either. It definitely sounds like a bug though