olofk/fusesoc #394: Addings support for new backed requires changes in both edalize and fusesoc

tudortimi, 2020-04-30:
While trying to add support for Xcelium, I made the changes I needed to edalize. I was trying it out in the FIFO repo (which uses CAPI1), but no source files were getting passed to the backend. I narrowed it down to `capi/core.py`, where there is the following snippet:

```
def _get_flow(self, flags):
        flow = None
        if "tool" in flags:
            if flags["tool"] in [
                "ghdl",
                "icarus",
                "isim",
                "modelsim",
                "rivierapro",
                "xsim",
                "vcs",
            ]:
                flow = "sim"
```

An entry for `xcelium` has to be added.

This had me confused for a big chunk of time, as I expected only the modification in edalize to be necessary. Release-wise it's also trickier to handle.

I'd propose adding something like a `flow` property to edalize backends, which fusesoc could interrogate and use accordingly, thereby making the addition of new backends easier.

tudortimi, 2020-04-30:
I also saw that there there is a plan to remove CAPI1 support.

olofk, 2020-05-27:
Yes, CAPI1 is deprecated and no new tool support will be added. CAPI1 cores should really have been migrated to CAPI2 long ago. Closing this

closed 2020-05-27 by olofk