olofk/fusesoc #396: Need a way to specify exported SystemVerilog headers

tudortimi, 2020-05-04:
The only thing I could find include-wise in the CAPI is that it's possible to specify which files are include files and the tool will set the parent directory of that file as an include directory (I think).

What's missing is the fact that a library might want to export file headers to be used by other libraries. This is probably not the case for designs, but TB libraries might want to provide some `cool_library_macros.svh` file that can be included in other libraries that depend on `cool_library`. This means that when compiling such dependents, they should get the parent directory of `cool_library_macros.svh` as an include directory.

(I used the term library instead of core, because it sounds strange to say I have a TB core.)

olofk, 2020-05-04:
I have a feeling that I haven't understood the issue correctly.

If you tag `cool_library_macros.svh` with `is_include_file : true`, then the directory containing the file with be automatically added the the list of include directories, exactly as you describe.

If another core/library wants to use `cool_library_macros.svh`, it just needs to make sure it has a dependency on the core/library that defined `cool_library_macros.svh`

I'm not that experienced with SV though, so I might be completely missing the point

imphil, 2020-05-04:
See https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/prim_assert.core#L11 for an example of what @olofk mentioned.

tudortimi, 2020-05-04:
@olofk You didn't misunderstand. That's what I meant.

closed 2020-05-04 by tudortimi