olofk/fusesoc #397: Need way to specify include path directly

tudortimi, 2020-05-04:
The only thing I could find include-wise in the CAPI is that it's possible to specify which files are include files and the tool will set the parent directory of that file as an include directory.

This assumes that files are always included "directly". It's also common to include files using relative paths containing directories:

```
`include "some_file.svh"
`include "some_directory/some_other_file.svh"
`include "some_other_directory/yet_another_file.svh"
```

imphil, 2020-05-04:
Thanks for your use case. Do you have an idea how to implement that?

With the current implementation to extract the path from files specified with `is_include_file`, FuseSoC can guarantee that all include files are actually known to and tracked by fusesoc. That's important once we move to "no-op builds", where builds are only executed if input files change. If we'd allow users to specify an arbitrary string as "include directory", which is then passed to the tool, this guarantee would be lost. 

Can you think of a way to get the best of both worlds?

tudortimi, 2020-05-04:
I imagine we could add some extra property for include files, like `include_path`, which per default is the parent directory of the file, but if specified is what gets used. Some checking that it's a relative path, accessible from the root would also be necessary.

tudortimi, 2020-05-15:
Any input on adding the new property to include files?

tudortimi, 2020-05-15:
Also, as I've stated elsewhere, this feature is probably mostly useful for verification code. Without #367, though, I can't really see myself using FuseSoC to manage TBs with hundreds of files,

olofk, 2020-05-15:
This is something that comes up from time to time. I am currently packaging code which does `include dir/file.vh`. As you have noticed, it's pretty awkward to do that in FuseSoC (I am too embarassed to show the [workaround](https://github.com/fusesoc/fusesoc-cores/blob/master/pulp-platform.org/include/dummy) I [did](https://github.com/fusesoc/fusesoc-cores/blob/master/pulp-platform.org/common_cells-1.16.4.core#L9))

We have previously discussed adding an `include_path` option (should be a bug report about that somewhere) but never decided if this should be per fileset or per target.

I had not thought about seeing it as a property per file, that is default set to `.`. This is actually quite nice because we avoid adding the dir of the include file in cases where it's not going to be used anyway.

The only case that is not covered would be if we need to add an include directory without having any files, but this is pretty far-fetched so let's forget that.

Now, technically, the way to do this would be to first add this property to the EDAM API, which is defined in Edalize and then modify the [helper function](https://github.com/olofk/edalize/blob/master/edalize/edatool.py#L260) that all backends use to get the list of files and include dirs.

Once that is in place, we can add it to CAPI2 and start using it from FuseSoC. Ideally there should be a new release of Edalize first, so that we can make FuseSoC depend on this version.

So let's start on the edalize side

olofk, 2020-05-27:
I have added the support now. @tudortimi , @wallento Could you please check if it works as intended. It solves the issue I had at least

tudortimi, 2020-06-07:
@olofk I tried it out and it adds `.` as an include directory. I have an example here: https://github.com/tudortimi/using-fusesoc-for-verification-code.git

Steps:

```
cd simple_ivc
fusesoc --cores-root . run simple_ivc
cat build/simple_ivc_0.1.0/default-xcelium/edalize_build_rtl.f
```

You'll see `.` as the second incdir. I would have expected only one incdir, ` ../src/simple_ivc_0.1.0/src/main/sv`.

olofk, 2020-06-16:
IIRC FuseSoC always adds `.` as an include dir because it needs to find some files in the work root. I don't think that's related to this issue. Is it a problem, or just an observation?

Jbalkind, 2020-06-16:
For what it's worth I was able to successfully use the include_path property last week and avoid any former "embarrassing" workarounds!

wallento, 2020-06-16:
Yes, I can confirm the same. Just tried it today and it works for my case. Thanks!

olofk, 2020-06-16:
Thanks. That's good to know. Closing this now

closed 2020-06-16 by olofk