olofk/fusesoc #398: Need way to differentiate between internal and exported include path

tudortimi, 2020-05-04:
Currently, if a file is marked as an include file, it will be present both when compiling its core and any dependents. It should be possible to specify whether an include file is only required for the compilation of its core or whether it should be available for dependents:

```
package some_package;

// exported header file containing macros that can be used in dependents
`include "some_package_macros.svh"

// internal file containing class part of the package
`include "some_class.svh"  

endclass
```

tudortimi, 2020-05-15:
I'm not sure what the etiquette is here. I'd have no issue in trying to come up with a pull request to implement the enhancement, but I'd like to know first if this is something you're open to and in what direction to go.

imphil, 2020-05-15:
A pull request is always fine, as is a proposal of how you'd implement it in this issue.

On the issue itself: I'm not quite sure I understand what you're looking for. Edalize, for most tools, synthesizes/compiles all files in one run, so they end up in a single compilation unit. If I understand your proposal, you're looking for a way to compile files individually, and then only specify the include directories you need for that specific step?

Can you elaborate a bit more what the benefit of that behavior would be, or which problem you're trying to solve?

tudortimi, 2020-05-15:
For the ModelSim backend, it uses multiple `vlog` and `vcom` calls, so it uses multiple compilation units. Incisive/Xcelium would also support the same thing.

imphil, 2020-05-15:
Yes, but don't rely on that -- how many compilation units are produced is currently an undocumented implementation detail in edalize. If we want to make it supported, we need to provide a way to switch between the two types of behavior, modify the backends, etc. 

But to get back to my original question: what problem are you trying to solve?

tudortimi, 2020-05-15:
Regarding what I'm trying to solve, you got it correctly, I want to be able to specify include directories on a (sorf ot) 'per-compilation unit' basis.

I say "sort of", because, while the ModelSim backend compiles each file in its own compilation unit, I would still give all files from a FuseSoC core the same include directories.

Example: 

'small_core' contains 'small_core_file0.sv' and 'small_core_file1.sv', ... there are 'some_core_internal_include_file0.svh', 'some_core_internal_include_file1.svh', ... and 'some_core_exported_include_file0.svh', 'some_core_exported_include_file1.svh', ... When compiling 'small_core', it would get the directory for the internal include files as an `incdir` and also the directory for the exported include files as an `incdir`, while compiling all the `sv` files.

'big_core' contains similar files, but also depends on 'small_core'. It would additionally get the directory for the exported include files of 'small_core' as an `incdir`, aside from its own compile options. This way, 'big_core' is only allowed to include the exported headers from 'small_core'.

tudortimi, 2020-05-15:
This isn't really a use case for designs, where I guess include files are seldomly used, but for verification one mostly writes code in included files.

There's a lot of cargo cult out there when it comes to included files (add include guards, even though files should only be included once; use library names as file prefixes to ensure that file names are unique; etc.), but a strict separation between what is internal and what should be exported helps avoid the "issues" such "best practices" aim to solve.

imphil, 2020-05-15:
OK, so supporting that needs first a notion of a compilation unit in fusesoc and edalize, something we don't have currently. Can you file a issue for that probably on the fusesoc side first, and outline your thoughts there how to express that in a CAPI2 file? For example, are we OK with having a hard-coded "1 compilation unit == 1 file rule"?

Once we have agreement on the CAPI2 syntax, we can then implement that down and expose it to the EDAM file. That's when the edalize work starts to get the backends implement that properly.

Once all of that is sorted, we can pick up this issue again.

tudortimi, 2020-05-15:
IMO compilation unit could be handled later, by treating such an `is_internal` annotation on include files as a hint for backends that support multi-step compile (e.g. ModelSim, Xcelium, etc.). Backends that don't support it could just as well ignore it initially. I only see it as being useful for verification code currently.

tudortimi, 2020-05-15:
In any case, without #367, I can't really see myself using FuseSoC to manage TBs with hundreds of files, where I would use the feature discussed here.

olofk, 2020-05-15:
This one is tricky, so let me just start with some long-term goals that are relevant to this.
1. The backends that support multi-file compilation units and single-file compilation units should have a switch to support both modes. The rivierapro backend recently  got support for doing that. I'm thinking also that there might even be a case for treating each core as a separate compilation unit, but let's get to that if a real need shows up.
2. We want to support partial compilation. I don't want to do any parsing of the HDL files to figure out the dependencies automatically. Other build systems to that and it's a fragile process that can cause a lot of frustration. This means that we will have to do it some other way. My current idea is to somehow (this part needs to be implemented) detect which files that have changed and then recompile a) all files in the same core that is listed before this file and b) all cores that depend on this core. This of course require the specific backend to support partial recompilation and that we enable support in each backend. I have taken the first step towards this by exposing the dependency tree in the EDAM file and adding a property for each file to indicate which core they belong to.

But currently none of this is used, and the backends just see a flat list of files. This means they have no chance of knowing which are their own private include files. I can definitely see the use of private include files. In fact, I believe in practice most include files are only relevant for the core that defines them. And if we want to do partial recompilation, we sure as hell want to keep down the number of global ones. So I'm ok with some property to mark an include file as private. The question is if we need other types of files also being able to be private. Should we do `is_private_include_file : true`, `scope : private`, `is_include_file : true_but_i_want_to_be_private`? Just want to make sure we're covering all relevant use cases by adding such option so that we don't have to add any hacks on top of it

tudortimi, 2020-05-16:
> We want to support partial compilation. [...] My current idea is to somehow (this part needs to be implemented) detect which files that have changed and then recompile a) all files in the same core that is listed before this file and b) all cores that depend on this core.

I'd say a simpler model start with would be: if any file in the core changes, recompile the entire core. For Verilog, there are a lot of tricky issues (mainly related to global scope) that can cause changes to propagate to previously compiled files, requiring them to be re-elaborated. Also, if any dependency changes, recompile. This is probably redundant, but also, if any imported header changes, recompile.

tudortimi, 2020-05-16:
> Should we do is_private_include_file : true, scope : private, is_include_file : true_but_i_want_to_be_private? Just want to make sure we're covering all relevant use cases by adding such option so that we don't have to add any hacks on top of it

I'm for the extra property, `scope`. This allows you to decide later to make all include files private by default (backward incompatible, but maybe not very disruptive if include files are used sparingly), without forcing users to change the properties of all include files. Concretely, files that were marked as `include_file` will still be include files. The ones that were supposed to be private will stay private. The few that were supposed to be exported will need to be marked with `scope: public`.

tudortimi, 2020-05-16:
I'd also like to point out that compilation unit has a few more implications than just which `incdirs` are present in the invocation. Given the following three files:

```
// macros.svh
`define FOO
```

```
// mod0.sv

`include "macros.svh"

module mod0;
`ifdef FOO
  bit has_foo;
`endif
endmodule
```

```
// mod1.sv

// macros.svh is NOT included here

module mod1;
`ifdef FOO
  bit has_foo;
`endif
endmodule
```

If the two files are compiled as a single compilation unit, with `mod0.sv` compiled before `mod1.sv`, then **both** modules will have a `has_foo` signal.

If the two files are compiled as a single compilation unit, with `mod1.sv` compiled before `mod0.sv`, then only `mod0` will have a `has_foo` signal.

If the two files are compiled as separate compilation units, in whatever order, then only `mod0` will have a `has_foo` signal.

Separate compilation units doesn't necessarily mean separate tool invocations. Tools have switches to configure them to treat each file as an own compilation unit.

Naturally, separate tool invocations will always mean separate compilation units.

olofk, 2020-06-16:
Right. So let's begin with adding a property to mark include files as private. There is still some road to cover, but it's one step along the way

tudortimi, 2020-12-23:
I'm dropping this issue, as I've moved to different things.

closed 2020-12-23 by tudortimi