olofk/fusesoc #401: Parameters from upper level core file

wallento, 2020-05-08:
When an IP block defines a parameter, in particular `vlogdefine`, it should be possible to set it from the target block of the toplevel design.
For example an IP block can define:
```yaml
  parameters:
    MYDEFINE:
      datatype: int
      paramtype: vlogdefine
      default: 8
```
The toplevel should be able to set this via its targets section:
```yaml
  targets:
    synth:
      parameters:
        - MYDEFINE=2
```
This currently does not work, error:

```
ERROR: Parameter 'MYDEFINE', requested by target 'synth', was not found
```
The workaround is to add a duplicate for this parameter in the parameters section:
```yaml
parameters:
  MYDEFINE:
    datatype: int
    paramtype: vlogdefine
```

It should be allowed to do this without this workaround.

olofk, 2020-05-08:
The root cause is that when the EDAM file is created, Edalize loops over all the cores in the flattened dependency tree. For each core, it calls `get_parameters`. This function has no knowledge of other cores. I'm currently looking at some different ways to fix this, but not sure about the best way forward yet

olofk, 2020-05-08:
I also realize we need to define one thing here. Should a core have access to a) all parameters defined by its dependencies or b) the parameters that the depdencies have explicitly listed in the default target?

olofk, 2020-05-08:
How about [this](https://github.com/olofk/fusesoc/tree/depparams) @wallento ?

wallento, 2020-05-08:
Thanks, will have a look.
About your question: I believe it should be able to access them all. This is only from the target that is actually executed, right? This should be able to overwrite default values set by every define in the current working set. What do you think, @imphil?

olofk, 2020-05-27:
FYI, it turned out to be far simpler to just expose the parameters that are defined in the default target. This is also consistent with how other things get exposed to other cores (except for generators, apparently) and makes sense also as some parameters would only be valid for certain targets.

olofk, 2020-05-27:
I pushed this to master now. Please reopen if you find any issues

closed 2020-05-27 by olofk