olofk/fusesoc #403: CAPI 2 VPI include and library directories

GCHQDeveloper560, 2020-06-03:
I am trying to get a VPI module working with FuseSoC that makes use of a library not installed in a standard location. Compiling the module therefore requires the use of `-I` and `-L` to point to the `/library/path/include` and `/library/path/lib`.

CAPI 1 supported an `include_dirs` field, but CAPI 2 does not. It constructs a similar value from . CAPI 2 constructs something similar from any include files explicitly specified in a fileset. However, since I believe absolute paths aren't supported the workaround of specifying `/library/path/include/inc1.h` for at least one file doesn't work.

Neither core format appears to support a way to specify `-L`.

Would it make sense to add `include_dirs` and `lib_dirs` for CAPI 2 or is there a better approach?

GCHQDeveloper560, 2020-06-03:
Actually the workaround for `include_dirs` does work even with absolute paths. It's specifying something like `/path/to/module.cpp` that causes a PyYAML explosion. So, adding a single include file to the fileset would work for `include_dirs` but doesn't solve handling `-L`

olofk, 2020-06-16:
Ok, so the `-I` case is taken care of? I remember having absolute paths as a use case when designing this, but can't remember all the details. Question then is what to do with `-L`. I don't have enough use cases to figure out if adding a lib_dirs entry would be enough or we will soon realize there's also a need to add this_dirs and that_dirs too.

Should we go for an entry for adding custom options instead? Maybe even a `build_command` instead to allow users to specify a separate build script or Makefile in case the build instructions get more complicated? But then we could perhaps use a generator instead? I don't know. It's tricky

johnathan-convertino-afrl, 2023-01-23:
I did preform a bit of a hack to get libraries to work in my case.
```
filesets:
   lib :
    files :
      - lib/C89_pthread_ring_buffer/libringBuffer.a : {file_type : user, copyto : libringBuffer.a}
vpi:
  bin_file_io_vpi:
    filesets : [src, header]
    libs : [ringBuffer -L., pthread]
```
Copied to the root of the icarus sim, and then just appended -L. at the end of the library name. Totally a hack way of doing it.