olofk/fusesoc #404: Enable support for generators which depend on generators

imphil, 2020-06-10:
This is a follow-up to #391.

With the code in #391, generators are run only once during the build process, and not again if new dependencies show up as result of a generator run. (This is consistent with what FuseSoC did before #391.)

This was done intentionally to avoid generator loops and to improve performance (each new generator run requires resolving the whole dependency tree again, which takes a couple of seconds).

The workaround is currently to add the dependencies to the core that calls the first generator, as done in https://github.com/lowRISC/opentitan/pull/2452.

As a solution, we have at least two options:
- Recursively call generators as new dependencies show up. This will require ways to ensure the dependency tree stabilizes and ways to break cycles (most likely). And we need to invest in the performance of the dependency solver.

- Call (each type of generator) only once, but keep the dependency tree fragment around for other cores to depend on.

Option 2 seems to sufficient and easier to me, and correct under the assumption that generators are idempotent.

olofk, 2021-02-26:
I think we need to collect all these cases of generated cores that dynamically adds to the dependency tree and see how they are best solved. My gut feeling is that many of them can be solved in different ways that makes the dependency resolution less fragile

olofk, 2025-02-21:
We decided to not allow features that modify the dependency-tree dynamically.

closed 2025-02-21 by olofk