olofk/fusesoc #406: No datatype for floating point parameters

DHWinterWolf, 2020-06-15:
I've just been working on adding some parameters to a FuseSoC core for a VHDL entity that has generics of type REAL. On checking the documentation at https://fusesoc.readthedocs.io/en/master/ref/capi2.html#parameter, it looks like there's nothing suitable for this.

Empirically, it seems fine to use a string type, as least with ModelSim. However, this might not be true for other simulations (I'm a bit limited with what I can try at present), and it's not entirely intuitive. It might be nice to have a type like "float" or "real" or similar that is able to map onto the VHDL REAL type.

GCHQDeveloper560, 2020-06-15:
There was discussion of this issue hidden in #383 along with a [link to a untested implementation](https://github.com/GCHQDeveloper560/fusesoc/tree/real_genparam). I haven't submitted a PR due to the lack of testing. Are you able to try it out?

DHWinterWolf, 2020-06-16:
Ah, interesting! I'll give that a go and see what happens.

DHWinterWolf, 2020-06-16:
Since #383 was there first, I'll close this issue as more or less a duplicate, and reply in that one.

closed 2020-06-16 by DHWinterWolf