olofk/fusesoc #408: Does hook need to exit before run?

m-kru, 2020-06-15:
I am trying to achieve some co-simulation with FuseSoc. I try to run Python code as `pre_run` hook, however the simulation hangs forever. It looks like hook needs to exit before the target is run. Can someone confirm this hypothesis? If this is true, is there any way to run hook in the background? I know that I can have pre and post hooks, just create new process in pre hook and kill it in post hook, but this is more work and files.

olofk, 2020-06-16:
Yes, the hook currently needs to exit first. When I needed a background job, I did just like you describe with a fork and kill. Maybe it could be a good feature to allow FuseSoC to run hooks in the background and kill them off when FuseSoC exits.

imphil, 2020-12-18:
I'm not convinced that this is something that FuseSoC can handle reliably in the general case. For specific use cases, using pre_run to start a job in the background, and another hook at the end to kill it should do the trick.

@olofk What do you think? Keep this open, or agree that we don't have a desire to implement such functionality in the near term?

olofk, 2021-02-26:
Let's close this one. The preferred way to handle this is to have the pre_run hook itself start a background process that the post_run hook can kill afterwards. I think it's too much work to add this functionality directly into fusesoc.

And I'm dead sure I had an example of doing exactly this but I have no idea what it was. Thought it was for jtag_vpi where a test client is started before the sim and connects through a socket to the testbench, but I can't find any evidence. Maybe I just left it to die on an old computer somewhere

closed 2021-02-26 by olofk

m-kru, 2021-04-28:
@imphil What did you mean by writing "handle  reliably in the general case"?