olofk/fusesoc #414: Best practice for default target in dependencies?

Jbalkind, 2020-06-22:
I'm wondering what the best practice is for dealing with the restriction that only dependencies' default target can be used. If I have multiple tool- and flag-based configurations that I need to support, how is best to do that?

In particular, if I want to use certain IP for FPGA that's different from sim, then if a core is a top level, I can just define different targets. However, if the core is a dependency, only the default target can be used.

Should I add a bunch of `tool_vivado?` checks to the filesets used? What if I want to make use of use flags too? This seems like I end up reimplementing my different targets together into a single mess for the default. I'm wondering whether this is what people are doing or whether they're instead sharding the dependent stuff into multiple further dependencies, one for each use case?

I don't think I've worded this very well but hopefully some of you have some input on this.

m-kru, 2020-06-22:
The best way is probably to add support for optionally defining target in the dependency.
You are not the only missing this feature. Somehow no one has ever added this.
It should be relatively easy to add, and there should not be problems with backward compatibility.

@olofk what do you think?

imphil, 2020-11-11:
I just stumbled across this issue, sorry for not replying earlier: There is a flag available matching the target name, `target_*`, e.g. `target_sim`. You can use this in downstream filesets to match against the target which is currently being executed.

```yaml
filesets:
  files_rtl:
    files:
      - rtl.sv
  files_sim:
    files:
      - sim.sv

targets:
  default:
    - files_rtl
    - "target_sim ? (files_sim)"
```

This setup is slightly fragile, as you effectively do a name matching against the target with no error checking, i.e. you need dependent core files which follow your target naming rules. But that's probably OK as long as you don't depend on a large number of external cores.

Jbalkind, 2020-11-12:
Hm that approach does strike me as a way of just reimplementing downstream target selection. If that's the common way of doing it then I feel like we should just have the ability to depend on a dependency's target directly. That would further enable using dependencies that you don't control and the ability to have an upstream target depend on a downstream target of a different name.

imphil, 2020-11-13:
The flag solution exists today and has its uses, but I agree, it also has significant downsides: it is global and can give rather surprising results if you don't control the whole dependency tree.

Something more explicit and local would be nice. What kind of syntax are you thinking about? Something like that:

```
# Currently not supported FuseSoC syntax to depend on a specific target:
filesets:
  files_rtl:
    depend:
      # depend on the default target of my:other:core
      - my:other:core
  files_sim:
    depend:
      # depend on the sim target in my:other:core
      - my:other:core[sim]
```

The problem with that is that the sim target of my:other:core is likely to also contain RTL files (since you'd want those if you call fusesoc --target=sim my:other:core). So instead, you'd probably need to have a new target within my:other:core that only contains simulation files for inheritance (e.g. a new `sim_only_files`) target?

Another solution could be to inherit filesets, and not targets ... I'm sure that opens another can of worms and @olofk has thought about it before ... 

(`[]` is likely to be a bad idea since it overlaps with arrays in YAML, but I can't think of something beautiful right now either.)

olofk, 2021-02-26:
Yes. I have had the solution in mind for a long time. The solution is local use flags, but the problem is that this is not something that is easily achieveable with the current dependency manager. It seems like @imphil is on the track already. The idea, for those unfamiliar, would be at the lowest level to e.g. define the default target like

```yaml
targets:
  default:
    filesets:
      - "fpga? (fpga_fileset)"
      - "sim? (sim_fileset)"
      - "!sim? (!fpga? (some_kind_of_default_fileset_maybe))"
```

In the level above you will then do just like @imphil wrote with `my:other:core[sim]`. It's actually not a problem to use square brackets as we need to treat the vlnv as a string anyway because yaml gets angry e.g. when it starts with >=

I should mention that I have had an idea for how to retrofit local use flags to the current dependency manager, but I'm not sure it works and it's most definitely not a great idea. But anyway, I think we could register virtual cores for all combinations of use flags and treat them as different cores. Right now I can't recall the details of how I thought it would work out though, so perhaps we better just leave it

imphil, 2021-02-26:
I have code around that prepares FuseSoC to do this kind of inheritance, I'll push it as part of the refactorings I'm currently working on once the 1.12 dust has settled a bit.

olofk, 2021-02-26:
Oh, really? That's fantastic. Related to that we also need to to a job to register use flags somehow

sebinho, 2023-08-25:
Hi Guys,

Was there any progress made on this topic? This is something I am also very interested in.
Thanks!