olofk/fusesoc #416: IndexError: list index out of range when used with GHDL

dominiksalvet, 2020-06-26:
Hello @olofk and FuseSoC community! :wave:

First thank you so much for creating this software — after years of using makefiles, I am so excited to actually start using FuseSoC! :heart: However, I have encountered an issue.

I have been following the [FuseSoC documentation](https://fusesoc.readthedocs.io/en/latest/index.html) and until reaching the [Creating a Core chapter](https://fusesoc.readthedocs.io/en/latest/user/tutorials/2-creating_a_core.html), everything was fine. Even that chapter was fine with Verilog and Icarus. However, as I currently develop my projects in VHDL with GHDL, I have decided to try the chapter with those two on a bit different design — a register file. Please note that everything is working while using GHDL manually.

Here are the source files:

<details><summary>registers.core</summary>

```yml
CAPI=2:
name: ::registers:0

filesets:
  rtl:
    files:
    - rtl/reg_file.vhdl: {filetype: vhdlSource}

  test:
    files:
    - test/reg_file_tb.vhdl: {filetype: vhdlSource}

targets:
  test:
    filesets: [rtl, test]
    toplevel: [reg_file_tb]
```
</details>

<details><summary>rtl/reg_file.vhdl</summary>

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity reg_file is
    port (
        i_clk: in std_ulogic;

        -- write port A
        i_a_we: in std_ulogic;
        i_a_index: in std_ulogic_vector(3 downto 0);
        i_a_data: in std_ulogic_vector(63 downto 0);

        -- read port B
        i_b_index: in std_ulogic_vector(3 downto 0);
        o_b_data: out std_ulogic_vector(63 downto 0);

        -- read port C
        i_c_index: in std_ulogic_vector(3 downto 0);
        o_c_data: out std_ulogic_vector(63 downto 0)
    );
end entity reg_file;

architecture rtl of reg_file is
    type t_registers is array (0 to 15) of std_ulogic_vector(63 downto 0);
    signal s_registers: t_registers;
begin

    -- asynchronous register read
    o_b_data <= s_registers(to_integer(unsigned(i_b_index)));
    o_c_data <= s_registers(to_integer(unsigned(i_c_index)));

    registers_write: process(i_clk)
    begin
        if rising_edge(i_clk) then
            if i_a_we = '1' then
                s_registers(to_integer(unsigned(i_a_index))) <= i_a_data;
            end if;
        end if;
    end process registers_write;

end architecture rtl;
```
</details>

<details><summary>test/reg_file_tb.vhdl</summary>

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity reg_file_tb is
end entity reg_file_tb;

architecture behavior of reg_file_tb is

    signal i_clk: std_ulogic := '0';

    -- write port A
    signal i_a_we: std_ulogic := '0';
    signal i_a_index: std_ulogic_vector(3 downto 0) := (others => '0');
    signal i_a_data: std_ulogic_vector(63 downto 0) := (others => '0');

    -- read port B
    signal i_b_index: std_ulogic_vector(3 downto 0) := (others => '0');
    signal o_b_data: std_ulogic_vector(63 downto 0);

    -- read port C
    signal i_c_index: std_ulogic_vector(3 downto 0) := (others => '0');
    signal o_c_data: std_ulogic_vector(63 downto 0);

    -- configuration
    constant c_CLK_PERIOD: time := 10 ns;
    shared variable v_done: boolean := false;

begin

    dut: entity work.reg_file
    port map (
        i_clk,
        i_a_we, i_a_index, i_a_data,
        i_b_index, o_b_data,
        i_c_index, o_c_data
    );

    clk_gen: process
    begin
        while not v_done loop
            i_clk <= '0'; wait for c_CLK_PERIOD / 2;
            i_clk <= '1'; wait for c_CLK_PERIOD / 2;
        end loop; wait;
    end process clk_gen;

    test: process
        variable v_data: unsigned(63 downto 0) := (3 downto 0 => '1', others => '0');
    begin
        wait for c_CLK_PERIOD;

        -- write to registers
        i_a_we <= '1';
        for i in 0 to 15 loop
            i_a_index <= std_ulogic_vector(to_unsigned(i, i_a_index'length));
            i_a_data <= std_ulogic_vector(shift_left(v_data, 4 * i));
            wait for c_CLK_PERIOD;
        end loop;
        i_a_we <= '0';

        -- read from registers
        for i in 0 to 15 loop
            i_b_index <= std_ulogic_vector(to_unsigned(i, i_b_index'length));
            i_c_index <= std_ulogic_vector(to_unsigned(15 - i, i_c_index'length));
            wait for c_CLK_PERIOD / 4; -- delta delay
            assert o_b_data = std_ulogic_vector(shift_left(v_data, 4 * i));
            assert o_c_data = std_ulogic_vector(shift_left(v_data, 4 * (15 - i)));
            wait for c_CLK_PERIOD - c_CLK_PERIOD / 4;
        end loop;

        v_done := true; wait;
    end process test;

end architecture behavior;
```
</details>

Now when I run `fusesoc run --target=test --tool=ghdl registers`, it ends up with the following error report (return value is 1):

<details><summary>Traceback</summary>

```
WARNING: Unknown item version in section Root
INFO: Preparing ::registers:0
INFO: Setting up project
WARNING: ../src/registers_0/rtl/reg_file.vhdl has unknown file type ''
WARNING: ../src/registers_0/test/reg_file_tb.vhdl has unknown file type ''
Traceback (most recent call last):
  File "/home/dominik/.local/bin/fusesoc", line 11, in <module>
    sys.exit(main())
  File "/home/dominik/.local/lib/python3.6/site-packages/fusesoc/main.py", line 836, in main
    args.func(cm, args)
  File "/home/dominik/.local/lib/python3.6/site-packages/fusesoc/main.py", line 369, in run
    args.build_root,
  File "/home/dominik/.local/lib/python3.6/site-packages/fusesoc/main.py", line 462, in run_backend
    backend.configure([])
  File "/home/dominik/.local/lib/python3.6/site-packages/edalize/edatool.py", line 112, in configure
    self.configure_main()
  File "/home/dominik/.local/lib/python3.6/site-packages/edalize/ghdl.py", line 109, in configure_main
    standard=stdarg[0].split("=")[1],
IndexError: list index out of range
```
</details>

> It creates the `build/registers_0/test-ghdl/Makefile` but it is empty. If I am not mistaken, it has something to do with the generation process of that makefile.

For the sake of completeness, the not-polished commands I use to manually invoke GDHL that work:

```sh
ghdl -a --workdir=work -Pwork --std=93 --ieee=standard rtl/reg_file.vhdl test/reg_file_tb.vhdl
ghdl -e --workdir=work -Pwork --std=93 --ieee=standard reg_file_tb
ghdl -r --workdir=work -Pwork --std=93 --ieee=standard reg_file_tb --vcd=graph.vcd
```

---

My versions of related software:

```
$ ghdl --version
GHDL 0.36-dev (v0.35-434-g534fd656) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
...

$ fusesoc --version
1.11.0

$ pip3 show edalize
Name: edalize
Version: 0.2.2
...

$ python --version
Python 2.7.17

$ python3 --version
Python 3.6.9
```

Please not that I use LLVM backend for GHDL, not mcode. Could that be the cause of the problem? Or is it something else? Thank you for your time. :rocket:

Jbalkind, 2020-06-26:
Could it be because you used `filetype` rather than `file_type` in your core? Seems that you used the former where I usually use the latter

ThomasHornschuh, 2020-06-26:
Hi, this is a known bug in edalize, which I addressed with PR  https://github.com/olofk/edalize/pull/131
Unfortunately I had not time yet to update the tests, so Olof was not able to integrate the PR into edalize. 
As workaround you can also set file_type to "vhdlSource-93" which effectively adds the std=93 parameter to ghdl.
Please note that there is currently no way to directly specify  93c in  a fusesoc core, this will only be used when you mix vhdlSource-87 and vhdlSource-93 in a project. But 93c is the default value for the std parameter of ghdl, and many existing vhdl sources (e.g. on OpenCores, or example code from Xilinx, etc.) require  --std=93c to sucessfully compile.
My  patch will just set std=93c explicitly when you specify file_type=vhdlSource

dominiksalvet, 2020-06-28:
Hello @Jbalkind and @ThomasHornschuh, sorry for the delay. You both are right — it is working now, thank you. :+1:

First, I have made a mistake with using `filetype` instead of `file_type` — my bad, sorry. And the second problem was with the `vhdlSource` and `vhdlSource-93` issue. It is funny that I had tried the `filetype` and `vhdlSource-93` combination before creating this issue. I was so close. :disappointed:

---

@ThomasHornschuh Hope it will get merged soon since it is very weird that the default GHDL configuration is not working.

closed 2020-06-28 by dominiksalvet