olofk/fusesoc #418: Option to provide git commitish to checkout for core library

GCHQDeveloper028, 2020-06-29:
When using the git provider for a Library of core files it will always use the default branch. I think being able to provide a commitish to checkout along with the URI is valuable to support these use cases:

- Consistent builds
  - It's useful to be able to rebuild a previous version of an FPGA image/run old simulations when debugging. To achieve this, fusesoc must resolve the dependencies the same way it did on the first build. If a library can be used at a specific commitish, it would be possible to provide fusesoc with the same view as the original build, thus giving the same build results. 
  - When building in a CI context it's important that the build is self consistent, it is possible that two parallel builds see slightly different dependency sets if a library changes during the build. If we fix the library to a specific commitish this can't happen
- Many package managers provide different 'streams' that one can follow, for example: nightly builds, stable, LTS, dev, int, rel etc. To do this in fusesoc at the moment would require a repository per-stream. If a commitish could be used, these can become branches on a single repository which, in my opinion, is much neater :)
 
Looking through the code I think there are two possible ways of implementing this

**Option 1**: Additional (optional) configuration item in fusesoc.conf, maybe `sync-version = <string>` which a provider can interpret as appropriate, for the current set of providers I think this only makes sense for git. It looks as though the `sync-uri` is only used to by git too so method this fits with that. This would require an update to the command line interface to allow a version to be provided.

Example config:
```
[library.fusesoc-cores]
location = /home/user/fusesoc-cores
sync-uri = http://a.git.repository/fpga/fusesoc-cores
sync-type = git
sync-version = nightly
auto-sync = true
```

**Option 2**: Use a special syntax and pass the version as part of the `sync-uri`. Pip does something along these lines by separating the URI and commitish with an `@`: `pip install git+https://github.com/olofk/fusesoc.git@1.11.0`. This would avoid adding another configuration option that only git uses and would mean no change to the command line interface for fusesoc. Correctly parsing such a string might be a little tricky, as some repositories use ssh and have a username also separated with an `@`: `git@github.com:olofk/fusesoc.git@1.11.0`. This option would only require changes to the git provider.

Example config:
```
[library.fusesoc-cores]
location = /home/user/fusesoc-cores
sync-uri = http://a.git.repository/fpga/fusesoc-cores@nightly
sync-type = gitauto-sync = true
```

I'll put together some changes to try option 1

GCHQDeveloper028, 2020-06-29:
Created pull request #419 implementing option 1 for comments

olofk, 2023-02-23:
So, the PR for this accidentally lost when I renamed the default branch. I managed to salvage it though and rebase it against the current main branch, so I'm happy to report this is implemented now. Thank you for the contribution

closed 2023-02-23 by olofk