olofk/fusesoc #420: Unknown item version in section Root

dominiksalvet, 2020-06-30:
Hello FuseSoC team! :wave:

As the issue title goes, whenever I invoke FuseSoC with a working `.core` file, I get the following warning as the first line of the output and then the normal output follows:

```
WARNING: Unknown item version in section Root
INFO: Preparing ::blinky:0
INFO: Setting up project
...
```

I basically followed the guide on the internet and right now I am in the state of the end of [this chapter](https://fusesoc.readthedocs.io/en/latest/user/tutorials/2-creating_a_core.html). It is annoying and I do not know what it means. Please, can you help me to eliminate the warning or at least explain what is its cause?

If you need more information, please let me know. :rocket:

tinylabs, 2020-07-14:
I see this too. Doesn't seem to affect anything but not sure where it's coming from. Even with a fresh install running 'fusesoc core-info fifo' produces this.

olofk, 2020-08-18:
Yeah, I see it sometimes too. It's harmless but should be fixed anyway. No clue where it comes from though. I'll look into it next time I come across it

olofk, 2020-08-18:
Fixed it now. It was a syntax error in jtag_vpi/jtag_vpi-r5.core in the base library

closed 2020-08-18 by olofk