olofk/fusesoc #421: Check-out of remote core on a per-repo basis ?

krishnan-gopal, 2020-07-01:
Hi

Is it possible to have fusesoc check-out remote cores without landing up with multiple copies of the same repository and version ?

Currently, fusesoc checks out a repository for every remote core even when their repo URL and version are exactly the same.
I understand that fusesoc has been implemented with a 'one core per repo' perspective, but such an enhancement could save a lot of space, when a repository has multiple cores.

imphil, 2020-07-01:
The behavior you're outlining is a good optimization, but it's currently not implemented. We'd appreciate a pull request, however!

krishnan-gopal, 2020-07-01:
Ok, I will look into this

olofk, 2021-02-26:
The problem with this arises when you have multiple cores and e.g. want version 1.2 of one core and 4.1 of another. Then you need to do multiple checkouts anyway and keep track of which version that corresponds to a certain core. I don't think the complexity is worth the savings in disk space and downloads.

What you probably want to do instead in these cases is to add the repo as a library with `fusesoc library add`

closed 2025-02-21 by olofk