olofk/fusesoc #426: Provider section: _ is removed from git tags with only numbers

krishnan-gopal, 2020-07-16:
In the CAPI2 provider section, if the version references a git tag with only numbers and the underscore _ fusesoc removes the _ while using it as a version for checkouts

This is the behavior I see when I use the provider section below:
```
provider:
  name : git
  repo : xyz
  version : 100_200_300
```

Output:
> INFO: Checking out xyz to <destination>
> DEBUG:     git clone -q xyz <destination>
> DEBUG:     git -C <destination> checkout -q 100200300
> error: pathspec '100200300' did not match any file(s) known to git
> ERROR: Failed to fetch '<CORE NAME>': "git -C <destination> checkout -q 100200300" exited with an error code. See stderr for details.

However, it works fine if the version contains an alphabet as shown:
```
provider:
  name : git
  repo : xyz
  version : 100_200_300_end
```

Output:
> INFO: Checking out xyz to <destination>
> DEBUG:     git clone -q xyz <destination>
> DEBUG:     git -C <destination> checkout -q 100_200_300_end

Jbalkind, 2020-07-16:
Does this still happen if you put quotes around the version number?

imphil, 2020-07-16:
Your `version` tag is parsed as integer by the YAML parser. You can avoid that by explicitly marking it as string with quotes around it. (See https://yaml.org/type/int.html for parsing rules.)

krishnan-gopal, 2020-07-16:
alright, that worked for me. Thanks!

closed 2020-07-16 by krishnan-gopal