olofk/fusesoc #428: Errors when running Marocchino test-bench using Fusesoc and Modelsim

0ena, 2020-07-26:
Hi,

I am following this tutorial: https://stffrdhrn.github.io/hardware/embedded/openrisc/2019/06/11/or1k_marocchino.html
and I am trying to run the simple Assembly example and produce the VCD file. Instead of opening the VCD with GTKwave I want to open it using Modelsim, so instead of Icarus as a tool, I and defining Modelsim, so that my command looks like the following:

`fusesoc run --target marocchino_tb --tool modelsim ::mor1kx-generic:1.1 --elf_load ./src/openrisc-asm --trace_enable --trace_to_screen --vcd` 

When I run it, I get the following errors:

```
user1@srv01:~/Proj/HW/fusesoc/momo$ fusesoc run --target marocchino_tb --tool modelsim ::mor1kx-generic:1.1 --elf_load ./src/openrisc-asm --trace_enable --trace_to_screen --vcd
WARNING: Unknown item version in section Root
INFO: Preparing ::adv_debug_sys:3.1.0-r1
INFO: Preparing ::cdc_utils:0.1
INFO: Preparing ::elf-loader:1.0.2
INFO: Preparing ::intgen:0
INFO: Preparing ::jtag_tap:1.13-r1
INFO: Preparing ::jtag_vpi:0-r2
INFO: Preparing ::or1k_marocchino:5.0-r3
INFO: Preparing ::uart16550:1.5.5-r1
INFO: Preparing ::verilog-arbiter:0-r3
INFO: Preparing ::vlog_tb_utils:1.1
INFO: Preparing ::wb_common:1.0.3
INFO: Preparing ::wb_bfm:1.2.1
INFO: Preparing ::wb_intercon:1.2.2-r1
INFO: Preparing ::wb_ram:1.1
INFO: Preparing ::mor1kx-generic:1.1
INFO: Setting up project

INFO: Running pre_build script check_libelf.sh
INFO: Building
/eda/tools/intel/Quartus19.1/modelsim_ase/bin/vsim -c -do "do edalize_main.tcl; exit"
Reading pref.tcl

# 10.5b

# do edalize_main.tcl
# ** Error: ../src/or1k_marocchino_5.0-r3/rtl/verilog/pfpu_marocchino/pfpu_marocchino_div.v(174): (vlog-2730) Undefined variable: 'qm_r'.
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/rtl/verilog/pfpu_marocchino/pfpu_marocchino_div.v(181): (vlog-2388) 'qm_r' already declared in this scope (srt4_fract58).
# ** Error: /eda/tools/intel/Quartus19.1/modelsim_ase/linuxaloem/vlog failed.
# Error in macro ./edalize_build_rtl.tcl line 50
# /eda/tools/intel/Quartus19.1/modelsim_ase/linuxaloem/vlog failed.
#     while executing
# "vlog +define+SIM +define+MODELSIM_SIM -timescale 1ns/1ps +incdir+../src/adv_debug_sys_3.1.0-r1/Hardware/adv_dbg_if/rtl/verilog +incdir+../src/or1k_mar..."
#  exit
cc -c -std=c99 -fPIC -fno-stack-protector -g -I/eda/tools/intel/Quartus19.1/modelsim_ase/bin/../include -I../src/elf-loader_1.0.2/ -o ../src/elf-loader_1.0.2/elf-loader.o ../src/elf-loader_1.0.2/elf-loader.c
cc -c -std=c99 -fPIC -fno-stack-protector -g -I/eda/tools/intel/Quartus19.1/modelsim_ase/bin/../include -I../src/elf-loader_1.0.2/ -o ../src/elf-loader_1.0.2/vpi_wrapper.o ../src/elf-loader_1.0.2/vpi_wrapper.c
ld -shared -E -o elf-loader_1.0.2 ../src/elf-loader_1.0.2/elf-loader.o ../src/elf-loader_1.0.2/vpi_wrapper.o -lelf
cc -c -std=c99 -fPIC -fno-stack-protector -g -I/eda/tools/intel/Quartus19.1/modelsim_ase/bin/../include  -o ../src/jtag_vpi_0-r2/jtag_vpi.o ../src/jtag_vpi_0-r2/jtag_vpi.c
ld -shared -E -o jtag_vpi_0-r2 ../src/jtag_vpi_0-r2/jtag_vpi.o 
INFO: Running
/eda/tools/intel/Quartus19.1/modelsim_ase/bin/vsim -do "run -all; quit -code [expr [coverage attribute -name TESTSTATUS -concise] >= 2 ? [coverage attribute -name TESTSTATUS -concise] : 0]; exit" -c -pli elf-loader_1.0.2 -pli jtag_vpi_0-r2 -L altera_mf_ver -L altera_mf -gpipeline=MAROCCHINO +elf_load=/home/user1/Proj/HW/fusesoc/momo/src/openrisc-asm +trace_enable=1 +trace_to_screen=1 +vcd=1 orpsoc_tb
Reading pref.tcl

# 10.5b

# vsim -do "run -all; quit -code [expr [coverage attribute -name TESTSTATUS -concise] >= 2 ? [coverage attribute -name TESTSTATUS -concise] : 0]; exit" -c -pli "elf-loader_1.0.2" -pli "jtag_vpi_0-r2" -L altera_mf_ver -L altera_mf "+elf_load=/home/user1/Proj/HW/fusesoc/momo/src/openrisc-asm" "+trace_enable=1" "+trace_to_screen=1" "+vcd=1" orpsoc_tb -gpipeline=MAROCCHINO 
# Start time: 17:45:43 on Jul 26,2020
# ** Error: (vsim-3170) Could not find 'orpsoc_tb'.
#         Searched libraries:
#             /eda/tools/intel/Quartus19.1/modelsim_ase/altera/verilog/altera_mf
#             /eda/tools/intel/Quartus19.1/modelsim_ase/altera/vhdl/altera_mf
#             /home/user1/Proj/HW/fusesoc/momo/build/mor1kx-generic_1.1/marocchino_tb-modelsim/work
# Error loading design
Error loading design
# End time: 17:45:43 on Jul 26,2020, Elapsed time: 0:00:00
# Errors: 1, Warnings: 0
Makefile:34: recipe for target 'run' failed
make: *** [run] Error 12
```

a) I suppose the qm_r error might be easy to fix, if I just move the definition of the "qm_r" reg a few lines higher in the Verilog code of "pfpu_marocchino_div.v". The problem is that I cannot find where Fusesoc gets this Verilog code, in order to make this change. From my understanding it gets it from an online repository every time I run the command, am I right on that? Any proposed workaround on this to fix the error?

b) Any clue on why the Makefile is not able to find the "orpsoc_tb"? Do I need to change any kind of path to point to it?

I have been challenging myself for quite a while in order to create a working SoC based on an OpenRISC processor for my De0-CV board (I started with OR1200) and I am kind of lost with the tutorials/broken links from open-cores/forks & repositories from different contributors. I keep finding new ones. :)
I see that this repo has new issues created/answered so I hope a response for mine also. If that is not the cases and it is no longer maintained, a confirmation would be greatly appreciated.

Kind regards,
Nassos

P.S. the work behind the OpenRISC/OpenCores/Fusesoc/or1ksim is immense! Keep it up!

bandvig, 2020-07-27:
Hi Nassos
If I understand correctly the error output you should be able to find FP-divisor source in `../src/or1k_marocchino_5.0-r3/rtl/verilog/pfpu_marocchino/pfpu_marocchino_div.v`  and the path is relative to the folder from which you run fusesoc.
It interesting to see that modelsim reports error for the `usage before declaration` case. Verilator, Icarus Verilog and Xilinx are OK with that. However, perhaps Xilinx could create a warning.
Currently I'm working on re-factoring MAROCCHINO FPU verilog. And of course I'll change position of qm_r declaration.

0ena, 2020-07-27:
Hi Andrey,

thank you for your reply. I managed to change the source file and move the "qm_r" declaration a bit higher and fix this error. But now I seem to get the following errors:

```
WARNING: Unknown item version in section Root
INFO: Preparing ::adv_debug_sys:3.1.0-r1
INFO: Preparing ::cdc_utils:0.1
INFO: Preparing ::elf-loader:1.0.2
INFO: Preparing ::intgen:0
INFO: Preparing ::jtag_tap:1.13-r1
INFO: Preparing ::jtag_vpi:0-r2
INFO: Preparing ::or1k_marocchino:5.0-r3
INFO: Preparing ::uart16550:1.5.5-r1
INFO: Preparing ::verilog-arbiter:0-r3
INFO: Preparing ::vlog_tb_utils:1.1
INFO: Preparing ::wb_common:1.0.3
INFO: Preparing ::wb_bfm:1.2.1
INFO: Preparing ::wb_intercon:1.2.2-r1
INFO: Preparing ::wb_ram:1.1
INFO: Preparing ::mor1kx-generic:1.1
INFO: Setting up project

INFO: Running pre_build script check_libelf.sh
INFO: Building
/eda/tools/intel/Quartus19.1/modelsim_ase/bin//vsim -c -do "do edalize_main.tcl; exit"
Reading pref.tcl

# 10.5b

# do edalize_main.tcl
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(93): (vlog-2163) Macro `OR1K_INSN_WIDTH is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(182): (vlog-2163) Macro `OR1K_SPR_SR_SM is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(182): near "]": syntax error, unexpected ']'.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(222): (vlog-2163) Macro `OR1K_OPCODE_SELECT is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(222): near "]": syntax error, unexpected ']'.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(222): (vlog-2163) Macro `OR1K_OPCODE_MFSPR is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(254): near "else": syntax error, unexpected else.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(254): (vlog-2163) Macro `OR1K_OPCODE_SELECT is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(254): (vlog-2163) Macro `OR1K_OPCODE_MTSPR is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(262): near "end": syntax error, unexpected end.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(263): (vlog-2163) Macro `OR1K_OPCODE_SELECT is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(263): (vlog-2163) Macro `OR1K_OPCODE_SD is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(264): (vlog-2163) Macro `OR1K_OPCODE_SELECT is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(264): (vlog-2163) Macro `OR1K_OPCODE_SH is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(272): near "end": syntax error, unexpected end.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(279): near "end": syntax error, unexpected end.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(291): near "end": syntax error, unexpected end.
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(295): (vlog-2388) 'insn' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(296): (vlog-2388) 'rA_num' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(297): (vlog-2388) 'rB_num' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(298): (vlog-2388) 'rD_num' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(299): (vlog-2388) 'rD_used' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(300): (vlog-2388) 'imm_16bit' already declared in this scope (or1k_marocchino_trace_print).
# ** Error (suppressible): ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(301): (vlog-2388) 'imm_26bit' already declared in this scope (or1k_marocchino_trace_print).
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(314): Illegal declaration after the statement near line '294'.  Declarations must precede statements.  Look for stray semicolons.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(324): (vlog-2163) Macro `OR1K_OPCODE_SD is undefined.
# ** Error: (vlog-13069) ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(324): near ")": syntax error, unexpected ')'.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(325): (vlog-2163) Macro `OR1K_OPCODE_SW is undefined.
# ** Error: ../src/or1k_marocchino_5.0-r3/bench/verilog/or1k_marocchino_monitor.v(326): (vlog-2163) Macro `OR1K_OPCODE_SB is undefined.
[... cut ...]
ld -shared -E -melf_i386 -o elf-loader_1.0.2 ../src/elf-loader_1.0.2/elf-loader.o ../src/elf-loader_1.0.2/vpi_wrapper.o -lelf
ld: skipping incompatible //usr/lib/x86_64-linux-gnu/libelf.so when searching for -lelf
ld: skipping incompatible //usr/lib/x86_64-linux-gnu/libelf.a when searching for -lelf
ld: cannot find -lelf
Makefile:52: recipe for target 'elf-loader_1.0.2' failed
make: *** [elf-loader_1.0.2] Error 1
ERROR: Failed to build ::mor1kx-generic:1.1 : 'make' exited with an error code
```

Any ideas on how to solve this would be much appreciated.
Thank you in advance for your response.

Kind regards,
Nassos

bandvig, 2020-07-27:
Good catch, Nassos.
We will add  `include "or1k_defines.v"` inside the "or1k_marocchino/bench/verilog/or1k_marocchino_monitor.v" soon.

Unfortunately, I can't help you with building elf-loader. It looks like libelf installed in your system is 64-bit version while build script tries to build 32-bit version of elf-loader (`-m32` option in `cc` command line). 
Perhaps @olofk could make a guide.

olofk, 2020-08-17:
The no-cost version of modelsim is problematic because it is 32-bit and no one has been running a 32-bit system in many years. This means that we need to have 32-bit versions of libelf and all its dependencies installed. I have stopped using the elf-loader core since it has thses problems and it also is pretty much impossible to compile on windows platforms. I think we should look at removing it from the marocchino SoC as well and load verilog hex files instead of loading elf files directly.

imphil, 2020-12-18:
@olofk Looks like there's nothing to be done on the FuseSoC side for this issue? Should we close it?

closed 2021-01-28 by imphil