olofk/fusesoc #432: Control automatic include directories

iabenz, 2020-07-28:
I am looking for a way to control which include directories are automatically added from files with the `is_include_file` attribute or to alternatively clear the entire set of include directories. I am seeing an issue with xsim where the include directories from the command line are incorrectly searched before the including file's directory when using the SystemVerilog `` `include "filename"`` directive. Although this is a tool-specific issue, it seems like it would be useful to have more granular control so that only the include directories that are actually required can be added.

imphil, 2020-07-28:
Thanks for opening this issue. I don't quite understand the motivation for this request, however. 

If a file is specified as `is_include_file`, it is intended to be used as such, and hence needs to be available in the include file path.

> I am seeing an issue with xsim where the include directories from the command line are incorrectly searched before the including file's directory

What do you mean by "incorrectly searched"? Can you describe the scenario and the problem a bit more?

iabenz, 2020-07-29:
Sure, I'll try to clear things up. The issue I'm facing is that the SystemVerilog project I am trying to migrate to fusesoc has duplicated header files in the tree. It looks something like this:
- directory1
  - inc.svh
  - source1.sv (`` `include "inc.svh"``)
- directory2
  - inc.svh
  - source2.sv (`` `include "inc.svh"``)

Now, this shouldn't be an issue since `` `include "inc.svh"`` (versus `` `include <inc.svh>``) should search in the current directory first and then optionally in other directories. It looks like xsim searches in the others directories first and then the current directory which can cause the wrong inc.svh file to be included for some directories. This was not an issue in the old build script since no include directories were added for xsim on the command line.

I know this is a lot of language and tool specific information, but I think it could be helpful, in general, to have more control over include directories. Maybe even just the option to specify what is needed manually.

olofk, 2020-08-17:
This was a tricky one. I'm not sure how to fix this long term. Do you mean that the problem disappear if you just avoid listing the include dirs explcitly? I trust that it's technically correct, but the setup is a bit niche so it would need more thinking to figure out the best way. In the meantime, I have some ideas for workarounds if you want to explore that. You could use the conditional syntax to avoid treating these files as include files when using xsim, e.g. something like

```yaml
filesets:
  f:
    files:
      - "tool_xsim? (inc.svh)"
      - "!tool_xsim? (inc.svh : {is_include_file : true})"
```

No idea if that will work though

iabenz, 2020-08-19:
Thanks for the workaround suggestions. I think using `file_type : user` is exactly the solution I needed. It allows me to have local include files for a core without having to rely on the `--no-export` option.

> Do you mean that the problem disappear if you just avoid listing the include dirs explcitly?

No, I was previously trying to solve my problem by explicitly removing the include dirs added by `is_include_file : true`.

olofk, 2021-02-26:
Hi @iabenz I'm wondering if we should close this one. The root cause has not really been fixed and there are certainly things to improve when it comes to include directories, but given that the issue was a tool bug and we found a workaround I think we could close this one and perhaps open another if we have more specific ideas for improvements in include directory handling

iabenz, 2021-02-27:
Hi @olofk, That sounds good to me. I will go ahead and close the issue.

closed 2021-02-27 by iabenz