olofk/fusesoc #434: Simulating with xcelium ignores dependencies

mtikekar, 2020-08-01:
I am a new user trying out the tutorial. I have both xcelium and icarus. When I ran `fusesoc run --target=sim i2c`, it ran as expected. But with `fusesoc run --target=sim --tool=xcelium i2c`, it errors out with a message:

```
instance 'tst_bench_top.vtu' of design_unit 'vlog_tb_utils' is unresolved in 'worklib.tst_bench_top:v
```

When I looked inside `build/i2c_1.14-r1/src`, I only see the i2c folder. (With icarus, I see i2c, vlog_tb_utils, wiredelay). And the corresponding files are missing from the list passed to the simulator.

The fact that even the source folders are missing led me to suspect that this is a bug in fusesoc and not in edalize. Or am I doing something wrong?

olofk, 2020-08-17:
Hi. I found the issue now. The wiredelay core was using the old CAPI1 format which required every supported simulator to be listed in the core description file. This is why FuseSoC didn't find any files. I have now added a CAPI2 version of wiredelay to the base library, so hopefully this should be fixed now if you run `fusesoc library update`. Please let me know if it worked or if any problem still persists

mtikekar, 2020-08-17:
Thanks! Now it added wiredelay but vlog_tb_utils is still missing. That also needs needs a CAPI2 version I guess.

Also, when I ran `fusesoc --verbose run ...`, it printed debug messages about wiredelay and vlog_tb_utils being CAPI1. Those should be fatal errors.

olofk, 2020-08-17:
You're absolutely right. I was so sure it was already CAPI2 that I didn't even check. I have released a new version now (::vlog_tb_utils:1.1-r1), so please do another `fusesoc library update` now and see if it works.

It's not really possible to make these fatal errors however, since it will work just fine in many cases. The correct thing to do is to convert everything to CAPI2. I have been incrementally doing that, but a few cores still remain

closed 2020-08-17 by mtikekar