olofk/fusesoc #435: Allow c++ source files in VPI + Allow custom compiler options to gcc

PeterCDMcLean, 2020-09-18:
Currently, the generated makefile only allows C files and no optional gcc compiler inputs (such as -x c++ to force C++ language).

The VPI fileset should allow and use cppSource file type tag to allow for C++ VPI.

And as an aside, I'm hacking VPI feature to enable DPI. I'm naming my VPI with .so extension and then using it in modelsim options :

```
vpi:
  dpic_axi.so: {filesets : [vpi_axi], libs: [c]}

...
  sim_axi:
    default_tool: modelsim
    filesets : [others_axi, sim_axi]
    toplevel : [test_axi_dpic]
    vpi : [dpic_axi.so]
    tools:
      modelsim:
        vlog_options: [-modelsimini ./modelsim_questa.ini -L xilinx_vip /tools/Xilinx/Vivado/2019.2/data/verilog/src/glbl.v]
        vcom_options: [-modelsimini ./modelsim_questa.ini -2008]
        vsim_options: [-modelsimini ./modelsim_questa.ini -t 1ps... **-sv_lib dpic_axi**]
```

The only caveat is that you have to create and link a vlog_startup_routines that is unused.
```
#include <vpi_user.h>
void (*vlog_startup_routines[])() = {
};
```

GCHQDeveloper560, 2020-09-25:
This is probably an issue more appropriate for the Edalize backend rather than FuseSoC. Does olofk/edalize#170 do what you need for C++ support? There's also an issue regarding DPI support (olofk/edalize#166) for xsim, though there's not been much discussion.

imphil, 2020-12-18:
Yeah, this is mostly a edalize question. @PeterCDMcLean please re-open if the pointers by @GCHQDeveloper560 are not sufficient and you need something else.

closed 2020-12-18 by imphil