olofk/fusesoc #438: Make parameters available as flag

imphil, 2020-11-11:
Make parameters (e.g. vlogdefine/vlogparam) available as use flag, to be able to do something like `depend: vlogdefine_SYNTHESIS ? corp:lib:synthesislib`.

imphil, 2021-01-22:
Turns out this is more involved. We only get the list of parameters and their values after we've finished resolving the dependency tree. So we need to do that first, and then redo the dependency resolution to take the newly found and set parameter flags into account. Or we find an easier solution that does the trick the most use cases ...

olofk, 2021-02-26:
I would do it the other way around and set the define depending on the value of the flag instead.
Flags = evaulated during core parsing
parameters/generics/defines = evaluated at compile-time
plusargs/cmdlineargs = evaluated at runtime

rswarbrick, 2022-03-02:
We've just run into another problem in OpenTitan where this would be really handy:

There's a module `A` in our design that may or may not instantiate a module, `B`, if some parameter, `P` is true. Module `B` exports some DPI functions. We've got a separate core, `C`, full of C++ code which uses those DPI functions. The code in `C` isn't actually run unless `P` is true.

The problem is that some tools (Verilator) completely throw away `B` and its DPI functions if `P` is false. In that case, it doesn't matter that `C` doesn't run: you still get a link error! We've been using a workaround which defines the DPI functions in `C` as well as weak symbols and checks at runtime that the code in `C` doesn't run if those symbols are null.

Unfortunately, this doesn't work with simulators like Xcelium. These compile the SV code's DPI exports into one `.so` library and the C++ code into a separate `.so` and then use `dlopen` to load them both at runtime. This doesn't obey weak symbol semantics and (it turns out) you end up with just the weak version of the symbol. Disaster!

This would be easy to fix if we could avoid depending on the cores for `B` and `C` when `P` was false.

olofk, 2022-03-09:
Hi @rswarbrick Could you use a flag `F` for this? instead of the parameter `P`? Flags can affect the dependency tree but parameters can't