olofk/fusesoc #448: Move FuseSoC configuration on Windows to APPDATA/LOCALAPPDATA

imphil, 2020-11-19:
Currently, FuseSoC stores its configuration file under Windows at `~/.config`, which resolves normally to something like `C:\Users\john\.config`. 

Until #447 we documented we'd put the config into LOCALAPPDATA, but actually put it into the home directory.

https://github.com/olofk/fusesoc/blob/24350b2d10a8b27819d84210554654baff6cd877/fusesoc/main.py#L203-L204

Using the home directory is very Unix-y, the more typical variant for Windows would be `%APPDATA%` (roaming, i.e. per user) or `%LOCALAPPDATA%` (per user and machine).

We should
* Decide on the use of LOCALAPPDATA vs APPDATA. (Where does pip place its packages? Do they roam?)
* Change the config file location.
* Include code to move the config file over if it is found in the old path.

olofk, 2020-11-21:
I'm not a big fan of moving sruff without a good reason since we need to add code for the migration parh and probably lose the backwards compatibility.

With that said, what does the xdg spec say about this on windows? ~/.config is the fallback if it can't find the xdg definitions. I' ve been considering to use the xdg_utils python package to save a few lines of code and add consistency but it has been low priority

imphil, 2020-11-21:
XDG isn't applicable to Windows at all, it's a Linux-only thing.