olofk/fusesoc #454: Drop Python 3.6 support in FuseSoC 2.0

imphil, 2020-12-17:
*In FuseSoC 2.0* we should drop support for Python 3.5, which has reached EOL in September 2020. 

One main benefit of doing so is significantly improved pathlib support (no more `str()` casts everywhere when using it), which will allow us to use more of its functionality when passing around filenames and paths in FuseSoC (which is currently typically done as string, and quite fragile at times).

The main downside is that we loose support for the "default Python" in Ubuntu 16.04 LTS, which will go out of standard support in April 2021 (which could match a FuseSoC 2.0 release quite well).

Jbalkind, 2020-12-18:
Title says "Drop Python 3.6" but you seem to mean "Drop Python 3.5" or "Require Python 3.6"?

imphil, 2020-12-18:
Oh yes, you're right. Fixed the title.

PR #460 (2021-03-18): Require Python 3.6
Python 3.5 is EOL upstream; remove support for it from FuseSoC to enable
us to use newer features.

Fixes #454
commit fe9d59b9 (2021-03-18): Require Python 3.6

Python 3.5 is EOL upstream; remove support for it from FuseSoC to enable
us to use newer features.

Fixes #454