olofk/fusesoc #455: Tracking: Changes to generators

imphil, 2020-12-17:
Currently, generators are only able to produce source files, which are then added to the file list. For advanced use cases, e.g. in OpenTitan, that's not enough. We currently are running a patched version of FuseSoC for OpenTitan and I'm working on changing a couple more things. This issue is tracking progress on that.

These changes will be changing the behavior of generators, and should therefor go into the next major release.

High priority:
- [ ] Resolve dependencies for generator-created cores: #391
- [ ] Pass information about discovered cores to a generator (EDAM file change): #250
- [ ] Allow dependencies to generator-produced cores: #449 

Lower priority:
- [ ] #382

olofk, 2023-01-19:
Is this still valid?

imphil, 2023-01-19:
Probably yes, there is no way to do what OpenTitan needs with current FuseSoC releases.

olofk, 2025-02-21:
We're dropping this as an alternative path was chosen