olofk/fusesoc #463: Correct way to only include testbench files for top level core file?

sjalloq, 2021-01-05:
That title is probably not much help in describing the issue I'm seeing so I'll expand a bit.  I have a number of core files in my hierarchy and have been using the conditional syntax for my default target.

```yaml
targets:
  default: &default_target
    filesets:
      - tool_verilator ? (files_verilator_waiver)
      - tool_modelsim ? (files_tb_sv)
      - tool_modelsim ? (files_tb_hex)
      - files_rtl
    toplevel: dut
```

However, now that I've expanded to two testbenches in my directory hierarchy, when I run the top level sim, I'm also seeing the sub-unit testbench compiled as the default target in the sub-unit core file also sees `tool_modelsim` being defined.

Is there a way to say something like `tool_modelsim && this.core ? (files_tb)` or am I going to have to revert the use of this conditional and use absolute filelists for my sim targets?  Seems like a common thing to want to do but I can't find any documentation.

Shareef.

imphil, 2021-01-06:
Generally: only put the reusable pieces of code (typically: RTL and lint waivers) into the default target. If you have a testbench for the IP core, put it into a separate target, e.g. a "sim" target, as shown in the example at https://fusesoc.readthedocs.io/en/master/user/build_system/core_files.html#ug-build-system-cb-blinky-full.

To help you with your specific problem, I'd need to see more of your two cores and the structure you're trying to establish.

sjalloq, 2021-01-06:
Thanks for this.  I ended up using `filesets_append` after finding a similar query in another issue.  That brings up another point which is the link you've pasted.  If I search for Fusesoc on Google, the first readthedocs.io page that hits must be an old version?  Does "rtd" mean retired?  https://fusesoc.readthedocs.io/en/rtd/tutorials/1-getting_started.html

I was wondering how a project that seems to have so much support had such bad documentation but your link seems to show that I was looking at the wrong one.  Can that page be deleted or at least updated with a big warning.

In terms of my issue, picture two levels of hierarchy, *top* and *subunit*, where *subunit* is instantiated within *top*.  I have a testbench for both where the core files both use the default target and the *tool_modelsim* conditional.  Compiling *tb_subunit* is fine because the core file has no dependencies and the tb files are added to the default target.  With the toplevel, *tb_top*, the core file has a dependency to the *subunit* which means that both sets of testbench files are compiled.

The subunit.core file looks like:

```yaml
files_rtl:
  files: [ subunit.sv ]
  file_type: systemVerilogSource

files_tb:
  files: [ tb_subunit.sv ]
  file_type: systemVerilogSource

default:
  filesets:
    - files_rtl
    - tool_modelsim ? (files_tb)
```

and the top.core file is:

```yaml
filesets:
  files_rtl:
    depend:
      - vendor:ip:subunit
    files: 
      - top.sv
    file_type: systemVerilogSource

  files_tb:
    files:
      - tb_top.sv
    file_type: systemVerilogSource

targets:
  default:
    filesets:
      - files_rtl
      - tool_modelsim ? (files_tb)
```
It's that combination of a dependency and the conditional that break things.

imphil, 2021-01-06:
> That brings up another point which is the link you've pasted. If I search for Fusesoc on Google, the first readthedocs.io page that hits must be an old version? Does "rtd" mean retired? fusesoc.readthedocs.io/en/rtd/tutorials/1-getting_started.html

Yes, that's a (very) old version of the documentation. I filed https://github.com/olofk/fusesoc/issues/464 to get it removed. [rtd means readthedocs, and it's probably an early testing branch for the documentation build system.]

imphil, 2021-01-07:
I don't know about your specific testbench structure, but generally, you don't inherit the testbench files, i.e. don't add them to the `default` target. That gives you the following files:

```yaml
# subunit.core
name: "vendor:ip:subunit"
filesets:
  files_rtl:
    files: [ subunit.sv ]
    file_type: systemVerilogSource

  files_tb:
    files: [ tb_subunit.sv ]
    file_type: systemVerilogSource

targets:
  # Only add reusable files into the default target, i.e. RTL files.
  default: &default_target
    filesets:
      - files_rtl

  sim:
    # "Inherit" all RTL files from the filesets directive in the default target
    <<: *default
    # Add your testbench files
    filesets_append:
      - files_tb
      # You can combine this with use flags if you have TB files which only a 
      # specific tool needs, e.g.:
      - tool_modelsim ? (files_tb_only_for_modelsim)
    # other keys, for example:
    toplevel: tb_subunit
    default_tool: modelsim
```

Now run `fusesoc --target=sim vendor:ip:subunit` to simulate the subunit with the default tool. Use `fusesoc --target=sim --tool=modelsim vendor:ip:subunit` to simulate with ModelSim explicitly. Without adding additional configuration, you can also pass any other supported tool to `--tool`, e.g. `--tool=vcs`.

The same goes for the `top.core`:

```
# top.core
name: "vendor:ip:top"
filesets:
  files_rtl:
    depend:
      - vendor:ip:subunit
    files: 
      - top.sv
    file_type: systemVerilogSource

  files_tb:
    files:
      - tb_top.sv
    file_type: systemVerilogSource

targets:
  default:
    filesets:
      - files_rtl

  sim:
    # "Inherit" all RTL files from the filesets directive in the default target
    <<: *default
    filesets_append:
      - files_tb
    # other keys, for example:
    toplevel: tb_top
    default_tool: modelsim
```

Again, run `fusesoc --target=sim vendor:ip:top` to simulate your toplevel.

imphil, 2021-01-07:
Btw, there is documentation on the default target at https://fusesoc.readthedocs.io/en/latest/user/build_system/core_files.html#the-default-target. Would you mind reading through it? Please let me know if there's something in there that confused you or things we could clarify. We probably need to strengthen the sentence "All reusable code in the core should go into the default target: RTL files, lint waivers, reusable constraints, etc." to explicitly say something like "do *not* put non-reusable (e.g. testbench) files into the default target".

sjalloq, 2021-01-07:
Hi,

thanks for this.  Yes, that all makes sense and is how I ended up fixing the issue.  The up to date documentation is indeed much more helpful  :-)  and I can't see anything in the default-target description that I think is missing.  Seems clear enough but being explicit can't hurt.  I've been thinking about the issue for a few days so if I came to this fresh I may read it differently.

As we're talking about the documentation, is there any support for running regressions?  Or has it just been designed to run single jobs?  What's the workflow if you've got a cluster or a bunch of cores and a load of tests to run?

imphil, 2021-01-07:
> As we're talking about the documentation, is there any support for running regressions? Or has it just been designed to run single jobs?

The primary focus of fusesoc is to configure and build a hardware design. The "run" support works well for "simple" use cases (on a developer machine, simple CI jobs, etc.), but it doesn't cover reaching out to a cluster, parallelizing jobs, etc. We have been thinking about that in the past, but found it very hard to find a "universal" solution; every company has their own setup to run regressions in a cluster (and collect results), and we'd somehow need to abstract away from that. 

Right now, the recommended approach is to use another tool or a CI system like Jenkins which first calls FuseSoC to configure and build the simulation binaries, and then uses the produced simulation binaries to run the regression with the right set of parameters. 

OpenTitan, for example, uses a custom tool called dvsim for running simulations with VCS (and many other things, so you'd need to carefully evaluate if that's the right tool for you), and its CI system to run Verilator simulations for different input sets.

Of course, if you have ideas how to build a reasonably generic regression functionality into FuseSoC (or an associated tool), we're happy to hear about your ideas!

sjalloq, 2021-01-08:
Thanks, I'll check out dvsim.

closed 2021-01-08 by sjalloq

olofk, 2021-02-28:
Late to the party, but for @sjalloq's original issue, I just wanted to mention that there is a built-in conditional called `is_toplevel` that I think does exactly what you're asking for. I use it like this for example https://github.com/olofk/serv/blob/master/serv.core#L32