olofk/fusesoc #464: Delete "rtd" branch/build on ReadTheDocs

imphil, 2021-01-06:
On readthedocs we have multiple versions of the docs, including an outdated "rtd" and a duplicated "master" version (https://readthedocs.org/projects/fusesoc/versions/). @olofk can you log into RTD and delete the rtd and master versions?

olofk, 2021-02-11:
Think I got it. Please confirm and close

imphil, 2021-04-13:
Hm, strangely https://fusesoc.readthedocs.io/en/rtd/tutorials/1-getting_started.html is still being served. I read through https://github.com/readthedocs/readthedocs.org/issues/1408 and it seems like previously RTD didn't fully delete the artifacts, but the issue reads to me that might be resolved now.

@olofk just to confirm: Even if you log in https://readthedocs.org/projects/fusesoc/versions/ doesn't show the "rtd" version any more?

(The problem is that the rtd URL is one of the first Google hits if you search for "fusesoc tutorial", and we don't want to point people to an outdated tutorial when they want to get started with FuseSoC.)

olofk, 2021-04-14:
Ouch. That sounds bad. I can't find any reference to the rtd and master branches in the docs so I have no idea what the problem is. Maybe just reach out to rtd and see if they know

imphil, 2021-05-11:
Filed https://github.com/readthedocs/readthedocs.org/issues/8174 upstream to ask for help.

imphil, 2021-05-14:
Looks good now, the branches are gone, and googling for `fusesoc tutorial` doesn't bring up the dead pages any more (at least for me).

closed 2021-05-14 by imphil