olofk/fusesoc #471: Can toplevel inherit name from the target name?

m-kru, 2021-01-17:
Almost always when I set `toplevel` within the target it has the same name as the target itself.
Example:
```yaml
  tb_start_after_reset:
    default_tool: ghdl
    toplevel: tb_start_after_reset
    filesets:
      - rtl
      - tb_start_after_reset
```

Is there any magic sequence (like `$`, `@`, `$1 `) to automatically assign the same name to the toplevel?
Example:
```yaml
  tb_start_after_reset:
    default_tool: ghdl
    toplevel: $
    filesets:
      - rtl
      - tb_start_after_reset
```

As you can see the same applies to one of the filesets.
The most compact version would be:
```yaml
  tb_start_after_reset:
    default_tool: ghdl
    toplevel: $
    filesets:
      - rtl
      - $
```

If it is not possible, are there any arguments against providing such functionality?

olofk, 2021-01-18:
I have been considering adding support for both user-defined (that can be set by use flags) and special variables (that are inferred by FuseSoC from the core description file). The use cases I have looked at would benefit from having name and version as special variables. I hadn't thought the target name, but that's a good idea. I don't think there's any yaml magic we can use for this so we have to do it ourselves. One thing that must be decided first is the syntax for variables.`$` doesn't work since we already support environment variables in core files. I have been considering %var% but I'm open to ideas. The important thing is that they don't clash with yaml syntax or other valid syntax in the core description files.

There's another approach to consider as well. I am normally very cautious when it comes to automatic things, but for toplevel, maybe we could consider setting it from the target name by default in case the user hasn't specified a toplevel themselves. Thoughts on that?

Also, related to this, there's a feature I introduced a while ago that might reduce clutter a bit if you have many targets like that. You can create a base testbench target with the common filesets, create an anchor and then use filesets_append, like this

```yaml
targets:
  tb_base: &base #This creates an anchor. The target only contain things that are common to all testbenches
    default_tool : ghdl
    filesets : [rtl]

  tb_start_after_reset:
    <<: *base #This brings in all the options from the tb_base target
    default_tool : modelsim  #Just an example if you want to override the default_tool from tb_base. Any options that is set here will override the same option in the tb_base target
    filesets_append : [tb_start_after_reset] #This appends the fileset the list of filesets that were inherited from the tb_base target
    toplevel : tb_start_after_reset
```

m-kru, 2021-01-19:
@olofk 
I like you idea about setting the `toplevel` by default to the target name much better than my initial proposal. 
It solves the case when this problem appears most often.
It looks like it would also be much easier and cleaner to implement.