olofk/fusesoc #474: Recommended way to setup Vivado step hooks

wfjm, 2021-01-23:
Vivado step hooks are essential for tailoring the default flows.  
The definition of a step hook names a script which is sourced when the step is reached.  
Therefore step hooks can't be included as TclSource in a fusesoc fileset, this will source the script at startup.

The trick via `copyto` given in #203 uses a CAPI1 feature and is thus obsolete.  

Question: what is the currently recommended way to setup Vivado step hook scripts ?

imphil, 2021-01-23:
`copyto` still works in CAPI2, you can find a working solution in the OpenTitan repository.

In short, it's a two-step process:

- First, produce a TCL file which sets up the hooks. This is just a normal TCL file added to your sources. 
    - https://github.com/lowRISC/opentitan/blob/810fcea0509bc81d91750f854754366e06345548/hw/top_earlgrey/top_earlgrey_nexysvideo.core#L27
	- https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/util/vivado_setup_hooks.tcl
- Second, add the files you want to run as hooks with `copyto`.
	- https://github.com/lowRISC/opentitan/blob/810fcea0509bc81d91750f854754366e06345548/hw/top_earlgrey/top_earlgrey_nexysvideo.core#L30
	- https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/util/vivado_hook_synth_design_pre.tcl

imphil, 2021-01-28:
@wfjm I'm closing this issue for now, but please let us know if the answer wasn't sufficient.

closed 2021-01-28 by imphil