olofk/fusesoc #478: Tutorial links broken

joshtyler, 2021-01-27:
The tutorial links in the main README are broken. I would submit a PR to fix, but I'm not sure exactly where you would like it to point! (maybe https://fusesoc.readthedocs.io/en/latest/user/tutorials/index.html ?)

There are two links that I can see:

"If you have one of the supported simulators installed, and want to do a quick check to see that it's working, follow the steps below, or look at the **tutorial** in the online documentation for a more thorough introduction."

"The online documentation contains a **tutorial** as well as information for users and developers of cores, or FuseSoC itself. For some quick communication with the active developers, feel free to join us at the FuseSoC chat. If you have found an issue, or want to know more about currently known problems, check out the issue tracker."

imphil, 2021-01-28:
Good point, thanks for letting us know! I'll have a look at fixing that.

imphil, 2021-01-28:
I opened #479 to address this and other issues with the README, could you have a look at it and let us know if you find areas where it could be improved?

joshtyler, 2021-01-28:
I've had a read through the reworked README in your PR, it looks good to me - nice and concise. Thanks for getting to this issue so quickly!

closed 2021-01-28 by joshtyler