olofk/fusesoc #488: How to parse filesets and run script to generate input file to target?

sjalloq, 2021-02-03:
I've been experimenting and have a number of questions regarding why things aren't working as I expect.  My ultimate aim was to understand how to run a pre-processing step before passing the output to a target.  So for example, running `sv2v` on a fileset to generate a single Verilog file to pass to Yosys.

1.  I can't even get a basic script running.  I can print my environment variables using `env` but if I try to run a script from a defined path, I get a not found.  In the following YAML if I add `env` to a target then it prints the environment variables include $PROJ_HOME but calling `sv2v` I get file not found.  Even a simple `ls $PROJ_HOME` can't see the environment variable.

```yaml
scripts:
  sv2v:
    cmd: ['python', '$PROJ_HOME/util/sv2vgen.py']
    filesets:
      - files_rtl
  env:
    cmd: ['env']
```

I've seen examples of copying scripts to the work directory using copyto.  Why can't arbitrary paths be specified?

2. How does a script access the filesets passed in?
3. If I'm generating an output file that feeds into another target, how are you supposed to specify that file as an input.  If it doesn't exist, the build phase falls over because it has a dependency on something that hasn't been generated yet.

Thanks.

imphil, 2021-02-03:
1. The command is not run through a shell or pre-processing to expand environment variables. Hence, you'll see a call to `python '$PROJ_HOME/util/sv2vgen.py'`. Either use a relative path to a file which is part of your fileset, or run the whole command through a shell (e.g. `/bin/sh -c 'python $PROJ_HOME/util/sv2vgen.py`).

> Why can't arbitrary paths be specified?

What do you consider an "arbitrary path"? Even though there are ways around it, cores are only really reusable if they are self-contained: using absolute paths or files which are not part of the core itself goes against that goal and isn't encouraged.

2. `files_rtl` in your example isn't passed in any way to the script.
3. Your "user" of the generated files needs to depend on the core that generated it; those dependencies are not handled on file-level, but on core-level.

All that said, doing sv2v pre-processing is rather tricky for a variety of reasons. You can find a working example at https://github.com/lowRISC/ibex/blob/abff9a886aa93fe7d402c0022aabe423632fea58/util/ibex_util_sv2v.core and https://github.com/lowRISC/ibex/blob/abff9a886aa93fe7d402c0022aabe423632fea58/util/sv2v_in_place.py (look in the `formal` directory of Ibex to try it out, you'll need to dig around a bit to see how it fits together).

sjalloq, 2021-02-03:
Thanks Phil, what's the point of the `filesets` within the `script` section?

sjalloq, 2021-02-03:
And in your `ibex_util_sv2v.core` file, what generates those `incdirs.txt` and `files.txt` lists?  Do you have another `core` file that does that?

sjalloq, 2021-02-04:
I've now got this working with a bit of a hack.  I'm using the scheme I mentioned above with the two core file approach.

My `sv2v.py` script reuses the FuseSoC front end to build the dependency tree and get a list of all include and source files.  These are then fed into `sv2v` which generates a single output Verilog file.

That single output file is written to the build directory where it overwrites a dummy file of the same name.  That dummy is there purely to stop FuseSoC falling over when it runs the setup stage.  It's just an empty `block_a.v` file which I can check in.

Probably not a very pure approach but unless anyone has any suggestions of features I've missed, then I'll stick with this.

olofk, 2021-02-05:
If it works for you that's good. Long-term I'm looking at reorganizing Edalize ([described in slightly more detail here](https://github.com/olofk/edalize/wiki/Edalize-(Slight-return))) to make some use-cases easier than what they are today. I _think_ your use-case would fit well into this, butt I'm not totally sure I got your needs right. Are you using FuseSoC/Edalize to run an EDA tool with the file that sv2v has created, or do you stop after generating it and do something else with it?

sjalloq, 2021-02-05:
Yes, so I'm running Yosys on the output of sv2v.  I'm trying to reuse FuseSoC as much as possible to avoid having separate VC files and/or flows outside of FuseSoC.  Ideally I'd like to use the FuseSoC core files as the golden dependency tree for all flows so right now as well as my sv2v script, I also use the front end to generate VC files for other tools.  That way the team doesn't need to keep two source tree lists up to date.

Going back to my original issue, I understood the reasons Phil gave regarding how each core file needs to be treated standalone but being able to pass arguments to script and/or have a pre_setup hook would make things a lot easier.  But then it only took me a day to pick up the basics and write a script that did what I needed.  I need to look at your proposal with fresh eyes to see how it might work for this use case.

closed 2021-02-09 by sjalloq