olofk/fusesoc #490: Allow custom file attributes

olofk, 2021-02-13:
Files support a few standard attributes like file_type and is_include_file but there are use cases thatI believe would be best served by custom attributes that are only valid for certain file types and tools

Examples would be Vivado constraint files where an attribute could decide if they are valid for synthesis or implementation. Or Libero where both Synplify and Libero itself can source tcl files but there's no way for Edalize to tell them apart. Maybe this can also be useful to add more flexibility to VPI/DPI. There have been some cases recently where the current VPI handling likely isn't flexible enough

abyszuk, 2021-10-04:
+1 to this, I'm relying heavily on module bound constraints, where you can for example bind a timing constraints file to a specific module. I think it's impossible to use them with current FuseSoc.
Example of such constraints is here: https://github.com/analogdevicesinc/hdl/tree/master/library/util_wfifo

javValverde, 2023-02-14:
I'm also interested in timing constraints scoped to a module. 
As an example on vivado:
`read_xdc -ref module <constraints.xdc>`

Is there a workaround for this?

olofk, 2025-02-21:
tags are supported since FuseSoC 2.2

closed 2025-02-21 by olofk