olofk/fusesoc #503: how to pass defines  in core file to .cc file

rahulraveendran15-coder, 2021-04-07:
Hi,

I just want to pass some defines from my command line to .core file and further this define has to pass to .cc file. This .cc file will create an output file and this output file name I need to control with the define name I am passing.
How can I do this? Does vlogdefine somewhat like below will work?

 RV32M:
    datatype: str
    default: ibex_pkg::RV32MFast
    paramtype: vlogdefine
    description: "RV32M implementation parameter enum. See the ibex_pkg::rv32m_e enum in ibex_pkg.sv for permitted values."

imphil, 2021-04-07:
To "pass" a define to a C++ file, you need to pass it to the C++ compiler (e.g. `gcc -DMY_DEF=value`). Since you're unlikely to call the C++ compiler directly, but probably as part of a Verilog simulator, you need to pass the define to the simulator in the way it expects it to be passed -- you need to have a look at the simulator manual for details. An example of how to do it in Verilator can be found at https://github.com/lowRISC/ibex/blob/master/dv/cs_registers/tb_cs_registers.core#L124.

Vlogdefine and similar FuseSoC parameters are not automatically passed as C defines, since that's completely tool-specific.

rahulraveendran15-coder, 2021-04-07:
Thank you,
is it possible to extract the value passing from command line like 
"fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --RV32Custom=1"

 I want to pass the value passed to --RV32M parameter along wih -D{RV32M}(this RV32M along with -D nes to get the value ibex_pkg::RV32MFast  ), is this possible?

imphil, 2021-04-07:
No, you cannot pass vlogparams/vlogdefines to the tool as "variable".

rahulraveendran15-coder, 2021-04-07:
Thank you very much. One more doubt regarding fusesoc is that. I am using tool as verilator so the corresponding code is as below
`   verilator:
        mode: cc
        verilator_options:
          - '-CFLAGS "-std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g -D"RV32M'
          - '-LDFLAGS "-pthread -lutil -lelf"'
          - "-Wall"
          - "-Wwarn-IMPERFECTSCH"
          - "-Wno-UNUSED"`
I know in this CFLAGS  section I can give my define -Dcustom_def=test.
But without disturbing the parameter(vlog and vdefine) is there any way to pass --custom_def=test from fusesoc command line to pass it to my .cc file?