olofk/fusesoc #505: Generating Board Specific IPs

hossein1387, 2021-04-08:
In my design, I am using Xilinx BRAM IPs that were generated for a specific FPGA board. In my core file, in the synthesis part, I am referencing this IP as below:

```
filesets:
    rtl:
           - ...
    synth:
        file_type : systemVerilogSource
        files:
           - rtl
            - ip/xilinx/bram_32Kb.xci:
                file_type: xci
    xdc:
        files:
            - synthesis/constraint_files/pito_core.xdc
        file_type: xdc

    synth:
        filesets:
            - synth
            - xdc
        default_tool: vivado
        tools:
          vivado:
            part: xcku040-fbva676-2-e
        toplevel: [rv32_core]
```

The `bram_32Kb.xci` was specifically generated for `xcku040-fbva676-2-e` FPGA. I want to test my design on different FPGAs. Is there a way in fusesoc to generate IPs depending on the part number?

olofk, 2021-04-08:
Xilinx IPs are a pain because of this reason and I generally recommend people to not use them. My recommedned option for an IP that is this simple would be to either write the functionality in pure Verilog so that you have something that is portable to all FPGA vendors and also easily runs in all simulators.

But if you want to use the Xilinx IP I would recommend using the TCL API instead of the xci files. You do that by creating a tcl file with the commands for generating the IP and then set file_type to tclSource. Here's an example of a TCL file to create an AXI CDC IP that I created for one of my designs

```tcl
create_ip -vlnv xilinx.com:ip:axi_clock_converter:2.1 -module_name axi_cdc -dir .
set_property -dict [list \
			CONFIG.DATA_WIDTH {64} \
			CONFIG.ID_WIDTH   {4} \
			CONFIG.ACLK_ASYNC {0} \
			CONFIG.ACLK_RATIO {1:4}] [get_ips axi_cdc]
```

To get the correct IP name and properties for your IP, you can use the Vivado GUI to generate the core and then check the TCL console to find the `create_ip`and `set_property` commands that were issued

imphil, 2021-04-08:
For BRAMs, there is no real reason to use ipgen, inference works perfectly well. (The story is different for things like MIG.) https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug901-vivado-synthesis.pdf gives you the right instantiation templates in Verilog and VHDL.

hossein1387, 2021-04-09:
thanks @olofk and @imphil I will try to follow you recommendations.

closed 2021-04-09 by hossein1387

davide-prandelli, 2024-07-26:
@olofk Do you know if there is a similar approach for Quartus? With Vivado it’s very simple, but with Quartus I’m really struggling to figure out how to do it. The idea is to create the IP core with tcl commands, so that I can include the tcl script in FuseSoC instead of the .qip file. I would be very grateful if you could help me.