olofk/fusesoc #507: design compiler support

timsaxe, 2021-05-19:
Olof,

I have added design compiler support.  Any interest in having it upstreamed? BTW, I was looking at how to add the documentation, and I can't find capi2.rst.  Is that expected?

imphil, 2021-05-19:
Design compiler support is great! I'd expect that most (all?) changes are required in the edalize backend? Can you open a pull request there?

> I can't find capi2.rst

capi2.rst is auto-generated from the source code in capi2/core.py and in edalize (for the backends).

Jbalkind, 2021-05-19:
Having a DC backend would be great! Do you just have it include a .tcl for all the user settings? Or is there a default tcl to drive it?

timsaxe, 2021-05-19:
I will open a PR in edalize.  I did not include a default tcl.  Not sure of the NDA status of the tcl file that I am using, so better safe than sued.

closed 2021-05-19 by timsaxe