olofk/fusesoc #508: Generate VCD files from xsim

paller, 2021-06-12:
It's possible to get VCD files out of xsim by adding a build target to the Edalize Makefile as

```Make
run-vcd: xsim.dir/$(TARGET)/xsimk
xsim -t xsim_script.tcl $(XSIM_OPTIONS) $(TARGET) $(EXTRA_OPTIONS)
```
with xsim_script.tcl containing

```tcl
open_vcd xsim_dump.vcd
log_vcd *
run all
close_vcd
quit
```
I'm trying to figure out if it's possible to get there within the possibilities of FuseSoc? I tried adding the tcl file to the target file list in the core file, but it then fails as it's handled as a verilog file.

paller, 2021-06-12:
I managed to get a VCD file generated with the following core definition but it's not pretty. Especially the xsim `-t` option sticks out.
```yaml
CAPI=2:
name: ::erv:0.0.1
description: Testing FuseSoc

filesets:
  rtl:
    files:
      - rtl/program_counter.sv
    file_type: systemVerilogSource

  tb:
    files:
      - test/program_counter_tb.sv
    file_type: systemVerilogSource

  xsim:
    files:
      - test/xsim_script.tcl

targets:
  default: &default
    filesets:
      - rtl
    toplevel: program_counter

  sim: &sim
    # Copy all key/value pairs from the "default" target.
    <<: *default
    description: Base simulation
    default_tool: xsim
    tools:
      xsim:
        xelab_options: [-debug typical]
        xsim_options: [-t ../src/erv_0.0.1/test/xsim_script.tcl]
    filesets_append:
      - tb
      - xsim
    toplevel: top

  sim_pc:
    <<: *sim
    description: Simulate the program counter
    toplevel: program_counter_tb
```
Would love to hear if there's a better way or if it's something worth making a pull request for.

GCHQDeveloper560, 2021-06-30:
I think this is something that would need to be added to the xsim Edalize backend as in olofk/edalize#127. It adds a `logged_hdl_objs` tool option that would let you list signals to be logged in VCD or SAIF. It's unfortunately out of date. I've not looked at the current state of these backends to see if the changes might be less invasive and easier to merge.

olofk, 2021-09-13:
One way to make it slightly less painful would be to do

```yaml
  xsim:
    files:
      - test/xsim_script.tcl : {copyto : xsim_script.tcl}
```

This will copy the tcl file to XSim's working directory so that you only need to reference it with `xsim_options: [-t xsim_script.tcl]`

But I also noticed now that the xsim backend doesn't handle tcl files natively. Ideally you should just need to specify `file_type : tclSource` for the tcl file and Edalize would make sure that XSim sources it automatically. This is the case for most backends but apparently not xsim