olofk/fusesoc #515: Unknown item 'Integer'. Parse Error

alperyazar, 2021-07-29:
Hi,

I am using FuseSoC with Vivado.  When I define a parameter with `Integer` type like `jtag_freq` or `jobs`, FuseSoC throws this error. Other ``String` type settings like `part` works fine.  Is it a problem related to CAPI2 parser, fusesoc/capi2/core.py ? It looks like parser doesn't know `Integer` type settings. FuseSoC 1.12.0 is installed with edalize 0.2.5. Problem exists both on Windows and Linux.

A question: Is it possible to override `jobs` parameter of Vivado (or any parameter in general) from command line when calling FuseSoC. I want to set this parameter on Linux with `$nproc` while calling the FuseSoC like `make -j$(nproc)`.

Thanks

imphil, 2021-07-29:
Thanks for your report. Can you show us a full core file that reproduces the problem?

alperyazar, 2021-07-30:
Thanks for the response. Here is the MWE.

`fusesoc.core`:

```yaml
CAPI=2:

name: a:b:c:0-r0

filesets:
  rtlverilog:
    files:
      - top.v
    file_type: verilogSource

targets:
  default:
    filesets : [rtlverilog]
  prj:
    default_tool: vivado
    filesets: [rtlverilog]
    toplevel : top
    tools:
      vivado:
        part: abc
#        jobs: 8
#        jtag_freq: 1
```

At the same level, created a dummy `top.v`. Then I run:

```bash
fusesoc library add test ./
```

then

```bash
fusesoc run --target prj --setup a:b:c
```

Everything works fine if `jobs` or `jtag_freq` is commented. When I uncomment at least one of them I get the error:

```text
WARNING: Parse error. Ignoring file xxx\fusesoc.core: Unkown item 'Integer'
```

Here is the installed packages with `pipdeptree -p fusesoc`

```text
$ pipdeptree -p fusesoc
fusesoc==1.12.0
  - edalize [required: >=0.2.3, installed: 0.2.5]
    - Jinja2 [required: >=2.11.3, installed: 2.11.3]
      - MarkupSafe [required: >=0.23, installed: 1.1.1]
  - ipyxact [required: >=0.2.3, installed: 0.2.4]
    - pyyaml [required: Any, installed: 5.4.1]
  - pyparsing [required: Any, installed: 2.4.7]
  - pyyaml [required: Any, installed: 5.4.1]
  - simplesat [required: >=0.8.0, installed: 0.8.2]
    - attrs [required: >=17.4.0, installed: 20.3.0]
    - okonomiyaki [required: >=0.16.6, installed: 1.3.0]
      - attrs [required: >=16.1.0, installed: 20.3.0]
      - jsonschema [required: >=2.5.1, installed: 3.2.0]
        - attrs [required: >=17.4.0, installed: 20.3.0]
        - pyrsistent [required: >=0.14.0, installed: 0.17.3]
        - setuptools [required: Any, installed: 49.2.1]
        - six [required: >=1.11.0, installed: 1.15.0]
      - six [required: >=1.9.0, installed: 1.15.0]
      - zipfile2 [required: >=0.0.12, installed: 0.0.12]
    - six [required: >=1.10.0, installed: 1.15.0]
```

Result is the same on both Windows and Linux. May be a incompatibility with latest version of `pyyaml` or `pyparsing`?

alperyazar, 2021-08-03:
Well, I focused on `capi2/core.py` [line 157](https://github.com/olofk/fusesoc/blob/324a2703799e458af29ce1eb239ccc3a08065aed/fusesoc/capi2/core.py#L157) and created following python script to test YAML loader with the same config

```python
import yaml
try:
    from yaml import CSafeDumper as YamlDumper
    from yaml import CSafeLoader as YamlLoader
except ImportError:
    from yaml import SafeDumper as YamlDumper
    from yaml import SafeLoader as YamlLoader

str_ = """
CAPI=2:

name: a:b:c:0-r0

filesets:
  rtlverilog:
    files:
      - top.v
    file_type: verilogSource

targets:
  default:
    filesets : [rtlverilog]
  prj:
    default_tool: vivado
    filesets: [rtlverilog]
    toplevel : top
    tools:
      vivado:
        part: abc
        jobs: 8

"""

try:
    dict_ = yaml.load(str_, Loader=YamlLoader)
except KeyError as e:
    raise SyntaxError(f"Unknown item {e}")

print (dict_)

```

and it worked, no exception. Interesting :thinking:

```text
{'CAPI=2': None, 'name': 'a:b:c:0-r0', 'filesets': {'rtlverilog': {'files': ['top.v'], 'file_type': 'verilogSource'}}, 'targets': {'default': {'filesets': ['rtlverilog']}, 'prj': {'default_tool': 'vivado', 'filesets': ['rtlverilog'], 'toplevel': 'top', 'tools': {'vivado': {'part': 'abc', 'jobs': 8}}}}}
```

closed 2021-09-13 by olofk

olofk, 2021-09-13:
To answer your original question, yes, you can run e.g. `fusesoc run --target=<mytarget> <mycore> --jobs=8`

If you run `fusesoc run --target=<mytarget> <mycore> --help` you will see all available options for that particular core and target

alperyazar, 2021-11-13:
Thank you @olofk . `--jobs=` does the job even it is not possible to set in `.yml` due the this error. This really speeds up on our CI server when a Vivado project has lots of OOC runs in it.

PR #516 (2021-09-13): Add Integer class for edalize
This commit add basic Integer class to support Integer type parameters for back ends defined in edalize. For example: jobs parameter of Vivado. Please see Issue #515 https://github.com/olofk/fusesoc/issues/515

Fixes #515