olofk/fusesoc #522: bazel backend for fusesoc

Dan10012, 2021-09-26:
Hi, 

During COVID lockdown people in my team ended up creating a POC of a build system similar to fusesoc, but using bazel.

I want to discuss the possibility to migrate fusesoc to use basel. For myself I'm unsure about pros & cons of the current python implementation, though I can elaborate on bazel (superficially, as I wasn't part of the group working on the build system).

Bazel is an open source build system by Google, designed to be language-agnostic, configurable to the specific use by rules (written in Starlark, a derivative of python).

Bazel brings to the table: 

- Sandboxing - building in a sandbox environmnet, where files / other env dependencies not explicitly specified can't be accessed. This prevents having hidden dependancies which can harm reproduceability.

- Transitive dependencies - Internal packages dependendy in a project (workspace, sort of equivalent to git repo) is transitive.  External dependencies are also supported, but they are not transitive (A depends on B depends on C might + A depends on D depends on C might lead to A depending on 2 different versions of C). So instead bazel demands explicit dependency of external transitive ones.

- Parallel Execution - bazel parallelizes independent build procceses, which leads to faster builds. (This can be useful with regressions for example)

- Smart Cache - bazel tracks changes in source files and together with the dependency graph it creats, it keeps tabs on what portions of the build proccess should be redone. This also results in faster builds.

This issue is mainly to discuss this possibility and pros & cons, as I'm really impressed with the fusesoc project but feel my technical knowledge to judge if the bazel option is good.

Regarding our implementation:
I can say that we are using the POC to simulate in vcs, generate modules, run regressions in CI, and working towards integrating it with our PnR python wrapper (which seems close to edalize). All configured similarly to how fusesoc does it (with equvalent to core files specifying targets etc...)

Unfortunately my workplace have a seperate network and it will be hard bringing out the bazel rules code (I'll have to ask not only legal premission but also security-wise...). There may be an early version of it using verilator that isn't in the seperate network. I'll see what I can do.

For the meantime you can look at [Bazel's documentation](https://docs.bazel.build/versions/4.2.1/bazel-overview.html) , and maybe in [this](https://github.com/hdl/bazel_rules_hdl)  example (which I'm unsure how good it is).

mithro, 2021-09-27:
Have you seen https://github.com/hdl/bazel_rules_hdl ?

mithro, 2021-09-27:
While Bazel has a lot of nice features, it is a very heavy dependency. I doubt it would be a good idea for FuseSoC to require it. However, (optional) FuseSoC and Bazel integration sounds like a good idea.

filmil, 2024-06-08:
Here you go. :)

https://github.com/filmil/bazel_rules_fusesoc

filmil, 2024-06-08:
In general, if you need to take a build tool into your bazel workflow, you usually add new rules.

olofk, 2024-06-10:
Very nice! Hope to find some time to look at your work.

I outlined quickly my own ideas for Bazel integration at Latch-Up earlier this year but haven't had time to implement it. https://award-winning.me/beyond-eda-lies-edalize/

filmil, 2024-06-15:
Thanks for your time in looking at this. I watched your video, but I am not sure how the bazel rules fit your vision - even though bazel was mentioned as one of the targets for generation.

Specifically there is a lot of overlap between what fusesoc does and what bazel can do. Bazel is also more general, which can be both good and bad, depending on how you look at it. Bad, because it's yet another tool that's vying for attention; and good if you co-design hardware and software and want a unified build system. (I also understand that typically this is not the case in hardware design world - it's usually you use whatever tools happen to be there.)
 
I also considered an automated fusesoc-to-bazel project converter, and while I actually made some progress there and that it should in principle be possible (though involved), I realized halfway that conversion is not the way.

Ideally bazel would absorb fusesoc through rules, but I think some fusesoc design decisions are making this challenging. The above fusesoc rules worked well for my purpose, which was to take in a fusesoc-described design wholesale. 

While this is suboptimal for me, I understand that using bazel in a hw toolchain is a curiosity and requires major work to pull off; while fusesoc is here today and it works.

In case you find this useful.

Cheers,
F

filmil, 2024-06-15:
> ...and maybe in [this](https://github.com/hdl/bazel_rules_hdl) example I've found today (which I'm unsure how good it is).

I wondered that too. They couldn't even be bothered to produce a working example, so I had to figure it out.

See here: https://github.com/hdl/bazel_rules_hdl/issues/123#issuecomment-1434405541

This was more than a year ago. Today, not much has changed in that respect. I suppose it's a rules repository "if you know, you know". Definitely does *not* meet the bar for reuse.

olofk, 2024-06-25:
Edalize has all the information about how to compile the command line for calling the various EDA tools with the correct parameters that the user has requested. Compiling a tool-specific command line is kind of what Bazel rules are (at least in my understanding). Currently we use this information to generate a makefile which we then call to run the flow. So, my idea is that instead of generating a makefile, we generate bazel build rules that describes how to call the EDA tools , and then it is up to the overlying Bazel flow to pick up these build rules and use as part of the flow (which I assume does more things than just invoking EDA tools, like compiling software etc). Does that make any sense?

filmil, 2024-06-27:
Hm, the build generation tooling makes quite a bit of sense to me. Automated build generation is already used in `bazel` tooling such as [`gazelle`][gz].

It just never occurred to me to base it on Edalize itself. In the past, I toyed with automated conversion of `core` files into bazel, and even got it to work to an extent. I didn't really go into all the detail and abandoned the approach, realizing very quickly that there is far too much detail for any external generation making sense in the long run. That seemed unsustainable.  An approach that leans onto `edalize` to generate bazel `BUILD` files would probably the best approach one could hope for.

That said, such build rules would still need to figure out how to make the toolchain work out of the box.  I strongly prefer having repository rules that allow bazel to set up a work repository from scratch without manual intervention or pre-installation from the user. (Versus system-wide installations, which make builds far less reproducible.)

[gz]: https://github.com/bazelbuild/bazel-gazelle