olofk/fusesoc #528: CI is currently broken on Windows

umarcor, 2021-11-04:
See, for instance, https://github.com/umarcor/fusesoc/actions/runs/1417056062.

imphil, 2021-11-08:
```
tox: py3-ci
  py3-ci installdeps: pytest, pytest-github-actions-annotate-failures
  py3-ci inst: D:\a\fusesoc\fusesoc\.tox\.tmp\package\1\fusesoc-0.1.dev1+g1576bac.zip
  py3-ci installed: atomicwrites==1.4.0,attrs==21.2.0,colorama==0.4.4,edalize==0.2.5,fusesoc @ file:///D:/a/fusesoc/fusesoc/.tox/.tmp/package/1/fusesoc-0.1.dev1%2Bg1576bac.zip,importlib-metadata==4.8.1,iniconfig==1.1.1,Jinja2==3.0.2,jsonschema==4.1.2,MarkupSafe==2.0.1,okonomiyaki==1.3.2,packaging==21.2,pluggy==1.0.0,py==1.10.0,pyparsing==2.4.7,pyrsistent==0.18.0,pytest==6.2.5,pytest-github-actions-annotate-failures==0.1.5,PyYAML==6.0,simplesat==0.8.2,six==1.16.0,toml==0.10.2,typing-extensions==3.10.0.2,zipfile2==0.0.12,zipp==3.6.0
  py3-ci run-test-pre: PYTHONHASHSEED='763'
  py3-ci run-test: commands[0] | /bin/rm -rf 'D:\a\fusesoc\fusesoc\.tox/.tmp/homedir'
  ERROR: InvocationError for command /bin/rm -rf 'D:\a\fusesoc\fusesoc\.tox/.tmp/homedir' (exited with code 2)
```

umarcor, 2021-11-08:
That seems to be coming from the `tox.ini`: https://github.com/olofk/fusesoc/blob/fa33702625895babde31d32ee84d78ed734bafa3/tox.ini#L16

olofk, 2021-11-08:
I guess the proper fix is to install a library in a throwaway dir for tests that need them. Or make them work with a mock library that the test suite supplies. Can't tell without digging deeper into this