olofk/fusesoc #529: parameter datatype file: missing path prefix

nopeslide, 2021-11-12:
According to the [parameter CAPI2 reference](https://fusesoc.readthedocs.io/en/stable/ref/capi2.html#parameter) the value of a `file` datatype should be prefixed with the path fusesoc in which was started.
I have a firmware file that needs to be loaded on simulation start:
```yaml
...
targets:
  ...
  jump_loop_tb:
    ...
    parameters:
      - g_FILE_PATH=firmware/jump_loop.mem

parameters:
  g_FILE_PATH:
    datatype: file
    description: Firmware file to load
    paramtype: generic
```
but in the resulting `*.eda.yml` files it is not prefixed:
```yaml

olofk, 2021-12-22:
Yes. I can see that the docs are a bit unclear about this. The `file` datatype is more intended for when you specify it as a parameter on the command-line, like `fusesoc run --target=some_target some_core --g_FILE_PATH=path/to/firmware.mem`

What you probably want to do is to use the `copyto` attribute for the firmware file, i.e. specify it as `- firmware/jump_loop.mem : {file_type : user, copyto : jump_loop.mem}` This will copy the file to a file called jump_loop.mem in the work directory which is where the EDA tool will be launched.

You can find an example of this [here](https://github.com/chipsalliance/Cores-SweRVolf/blob/master/swervolf.core#L37) where hello.vh uses copyto. A bit further down [here](https://github.com/chipsalliance/Cores-SweRVolf/blob/master/swervolf.core#L105) you can see that it can now be referred to using that path

olofk, 2022-12-18:
Is there anything else we want to do here, or can we close the issue?

nopeslide, 2022-12-19:
Thanks a lot for clearing this up for me and sorry I didn't answer.
I thought the docs could be made clearer at this point, but I'm also fine if this issue is closed